project-chip/connectedhomeip #15975: Wildcard Reporting with iOS chiptool fails -M5 temp updates do not reflect in chiptool

kean-apple, 2022-03-08:
#### Problem
Since switching to wildcard reporting for Chiptool, changing temp values on M5 board do not reflect in iOS chiptool

1. Pair M5 board using iOS chiptool2. 
2. Verify able to toggle light
3. Go to Temp sensor option, enable reporting Min: 1s, Max: 2s, Delta: 1C
4. Change temp on M5 board -> verify iOS chiptool updates

In step 4, Temp values do not update. See a bunch of errors reading values on M5 log

I (150619) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (150619) DeviceWithDisplay: editing attribute as integer: 34 (+)
E (150629) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to 771a7db8
I (150629) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (150649) app-devicecallbacks: Unhandled cluster ID: 1026
I (150659) app-devicecallbacks: Current free heap: 62824

E (150839) chip[IN]: Received a duplicate message with MessageCounter:16124935 on exchange 50158r
E (150839) chip[EM]: Received message of type 0x1 with protocolId (0, 1) and MessageCounter:16124935 on exchange 50158r
E (150859) chip[EM]: Found matching exchange: 50158r, Delegate: 0x3ffce1d8
E (150869) chip[EM]: Forcing tx of solitary ack for duplicate MessageCounter:16124935 on exchange 50158r
E (150869) chip[EM]: Sending Standalone Ack for MessageCounter:16124935 on exchange 50158r
E (150879) chip[IN]: Prepared secure message 0x3ffe7764 to 0xB2E6D3E8F9A5EAEC (1)  of type 0x10 and protocolId (0, 0) on exchange 50158r with MessageCounter:3760406.
E (150899) chip[IN]: Sending encryp
[reporting-fails-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8208989/reporting-fails-chiptool-logs.txt)
[reporting-fails-M5logs.txt](https://github.com/project-chip/connectedhomeip/files/8208991/reporting-fails-M5logs.txt)
M5 ted msg 0x3ffe7764 with MessageCounter:3760406 to 0xB2E6D3E8F9A5EAEC (1) at monotonic time: 150059 msec
I (150949) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (150949) DeviceWithDisplay: editing attribute as integer: 35 (+)
E (150959) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to 771a7db9
I (150959) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (150979) app-devicecallbacks: Unhandled cluster ID: 1026
I (150989) app-devicecallbacks: Current free heap: 62812

E (150989) chip[EM]: Retransmitting MessageCounter:3760404 on exchange 50158r Send Cnt 0
E (150999) chip[IN]: Sending encrypted msg 0x3ffca04c with MessageCounter:3760404 to 0xB2E6D3E8F9A5EAEC (1) at monotonic time: 150163 msec
E (151139) chip[IN]: Received a duplicate message with MessageCounter:16124935 on exchange 50158r
E (151149) chip[EM]: Received message of type 0x1 with protocolId (0, 1) and MessageCounter:16124935 on exchange 50158r
E (151159) chip[EM]: Found matching exchange: 50158r, Delegate: 0x3ffce1d8
E (151169) chip[EM]: Forcing tx of solitary ack for duplicate MessageCounter:16124935 on exchange 50158r
E (151169) chip[EM]: Sending Standalone Ack for MessageCounter:16124935 on exchange 50158r
E (151179) chip[IN]: Prepared secure message 0x3ffe7764 to 0xB2E6D3E8F9A5EAEC (1)  of type 0x10 and protocolId (0, 0) on exchange 50158r with MessageCounter:3760407.
E (151199) chip[IN]: Sending encrypted msg 0x3ffe7764 with MessageCounter:3760407 to 0xB2E6D3E8F9A5EAEC (1) at monotonic time: 150359 msec
E (152999) chip[EM]: Retransmitting MessageCounter:3760404 on exchange 50158r Send Cnt 1
E (152999) chip[IN]: Sending encrypted msg 0x3ffca04c with MessageCounter:3760404 to 0xB2E6D3E8F9A5EAEC (1) at monotonic time: 152164 msec
I (153089) ROUTE_HOOK: Received RIO
I (153089) ROUTE_HOOK: prefix FDB7:92FF:3249:: lifetime 1800

E (153189) chip[EM]: Received message of type 0x1 with protocolId (0, 1) and MessageCounter:16124936 on exchange 50158r
E (153189) chip[EM]: Found matching exchange: 50158r, Delegate: 0x3ffce1d8
E (153209) chip[EM]: Rxd Ack; Removing MessageCounter:3760404 from Retrans Table on exchange 50158r
E (153219) chip[EM]: Removed CHIP MessageCounter:3760404 from RetransTable on exchange 50158r
E (153219) chip[DMG]: StatusResponseMessage =
E (153229) chip[DMG]: {
E (153229) chip[DMG]: 	Status = 0x0,
E (153229) chip[DMG]: 	InteractionModelRevision = 1
E (153239) chip[DMG]: }
E (153239) chip[IM]: Received status response, status is 0
E (153249) chip[DMG]: <RE> OnReportConfirm: NumReports = 0
E (153259) chip[DMG]: IM RH moving to [GeneratingReports]
E (153259) chip[DMG]: <RE:Run> Cluster 3, Attribute 0 is dirty
E (153269) chip[DMG]: Reading attribute: Cluster=0x0000_0003 Endpoint=0 AttributeId=0x0000_0000 (expanded=1)
E (153279) chip[DMG]: AccessControl: checking f=1 a=c s=0xB2E6D3E8F9A5EAEC t= c=0x0000_0003 e=0 p=v
E (153289) chip[DMG]: <RE:Run> Cluster 3, Attribute 1 is dirty
E (153299) chip[DMG]: Reading attribute: Cluster=0x0000_0003 Endpoint=0 AttributeId=0x0000_0001 (expanded=1)
E (153309) chip[DMG]: AccessControl: checking f=1 a=c s=0xB2E6D3E8F9A5EAEC t= c=0x0000_0003 e=0 p=v
E (153309) chip[DMG]: <RE:Run> Cluster 3, Attribute fffd is dirty
E (153319) chip[DMG]: Reading attribute: Cluster=0x0000_0003 Endpoint=0 AttributeId=0x0000_FFFD (expanded=1)
E (153339) chip[DMG]: AccessControl: checking f=1 a=c s=0xB2E6D3E8F9A5EAEC t= c=0x0000_0003 e=0 p=v
E (153349) chip[DMG]: <RE:Run> Cluster 3, Attribute fff8 is dirty
E (153349) chip[DMG]: Reading attribute: Cluster=0x0000_0003 Endpoint=0 AttributeId=0x0000_FFF8 (expanded=1)
E (153359) chip[DMG]: AccessControl: checking f=1 a=c s=0xB2E6D3E8F9A5EAEC t= c=0x0000_0003 e=0 p=v
E (153369) chip[DMG]: <RE:Run> Cluster 3, Attribute fff9 is dirty
E (153379) chip[DMG]: Reading attribute: Cluster=0x0000_0003 Endpoint=0 AttributeId=0x0000_FFF9 (expanded=1)
E (153389) chip[DMG]: AccessControl: checking f=1 a=c s=0xB2E6D3E8F9A5EAEC t= c=0x0000_0003 e=0 p=v
E (153399) chip[DMG]: <RE:Run> Cluster 3, Attribute fffb is dirty

bzbarsky-apple, 2022-03-08:
Fixed by https://github.com/project-chip/connectedhomeip/pull/15943

bzbarsky-apple, 2022-03-08:
Hmm.  Maybe this was on a sha that includes that fix already... Reopening until that's clarified.

kean-apple, 2022-03-08:
this was tested on SHA: 2822511e632bd2ae8a2e5a853a114a9bac1cebe5

kean-apple, 2022-03-08:
New set of logs when settings reporting interval to 30
[enable-reporting-30s-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8209710/enable-reporting-30s-chiptool-logs.txt)
[enabling-subscriptions-30s-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8209714/enabling-subscriptions-30s-M5-logs.txt)
s

bzbarsky-apple, 2022-03-09:
That still seems to be in the middle of the priming read....

That said, I am seeing "Cluster 406, Attribute fff9" reported at least twice in the log, which seems pretty dubious.  @yunhanw-google can you take a look, please?

erjiaqing, 2022-03-09:
From the log, seems we are in the middle of a priming read, and the initial data takes long time to send, and is not finished at the end of the log.

This is expected if the data is reported later.

When the attribute was updated in the middle of a reporting, we will re-generate the report data given the fact that the attribute we are current reporting (might be a list) may be changed.

Seems list is the only thing we need to take care of since we cannot read it atomically between chunks.

The proposed solution is to regenerate the data only when the changed data is the one we are current reporting.

bzbarsky-apple, 2022-03-09:
> When the attribute was updated in the middle of a reporting, we will re-generate the report data given the fact that the attribute we are current reporting (might be a list) may be changed.

When _any_ attribute was changed we regenerate the entire priming report?  No one is changing the fff9 attribute on any clusters, I am quite sure.

erjiaqing, 2022-03-10:
> When _any_ attribute was changed we regenerate the entire priming report? No one is changing the fff9 attribute on any clusters, I am quite sure.

When we regenerate the entire report, we just reset the path iterator, then fff9 will be generated (another time)

In the log, we can find that we changed the "tempture" attribute in the middle of priming report, so it triggers a full regeneration.

bzbarsky-apple, 2022-03-10:
> In the log, we can find that we changed the "tempture" attribute in the middle of priming report, so it triggers a full regeneration.

Ah, ok.  @kean-apple are you changing things via the buttons on the m5stack before the priming read completes?

bzbarsky-apple, 2022-03-10:
But also, for a real temperature sensor that sounds like a huge problem, because it _will_ be updating the attribute continuously, right?  Even if it has some sort of throttle, nothing guarantees that throttle is to a rate slow enough that we can complete an entire "read everything" priming read.

@erjiaqing

kean-apple, 2022-03-10:
@bzbarsky-apple yes I was change buttons pretty quickly after enabling reporting since the previous non wild card reporting gets enabled much quicker

bzbarsky-apple, 2022-03-10:
OK.  @kean-apple can you try:
1. Set the 1s/2s reporting interval you were testing with initially.
2. Not touch the buttons until the priming read completes (probably around 15-20s, but may depend on network conditions and on how much the esp32 is logging).
3. Check whether using the buttons after the priming read completes works correctly.

mrjerryjohns, 2022-03-11:
I had initially thought we could avoid resetting the iterator during the priming read, and THEN process the dirty set after subscription establishment, but that runs into consistency issues if lists were modified/set dirty during priming and chunking was involved. That can result in us sending egregiously wrong list contents, and/or potentially having an invalid iterator into a list.

One option is to check if the iterator is pointing into a list that got modified by SetDirty and if so, just reset the iterator back to the front of the list, BUT avoid resetting the iterator in all other cases.

mrjerryjohns, 2022-03-11:
> The proposed solution is to regenerate the data only when the changed data is the one we are current reporting.

Didn't quite see this, but it lines up with what I said above.

mrjerryjohns, 2022-03-11:
@erjiaqing why do we need timestamps at all then in the proposed solution? The PR seems to indicate timestamps are involved..

kean-apple, 2022-03-11:
@bzbarsky-apple tried the steps you recommended, yes if I wait 30s for the priming reads complete and then change temp on the M5 I am seeing iOS chiptool update.

PR #16060 (2022-03-28): [IM] Record LastReportTick and DirtyTick in ReadHandler
#### Problem

Fixes #15975

During priming report, the report might be large, and if there are any attributes changed during the report, we will do a full regeneration, makes the report unable to finish.

#### Change overview
- Only regenerate the attributes under the current generating cluster (and only when needed)
- For any other attribute change, we record a timestamp, and report it the next time we are generating the reports.

#### Testing
- Updates existing tests for SetDirty and Regeneration
closed 2022-03-28 by erjiaqing