project-chip/connectedhomeip #15991: On/Off  Cluster:  command order issue in cmd `OffWithEffect`

helaywang-wiz, 2022-03-09:
I'm trying to implement `OffWithEffect` callback in the firmware app, but found below issue which confused me:
In `connectedhomeip/src/app/clusters/on-off-server/on-off-server.cpp` : `OnOffServer::offWithEffectCommand()`, it will firstly set OnOff attribute value and then trigger effect. So in the firmware app, the On/Off attribute change callback will be called before off effect triggered, my question is: Is this a expected behavior? is there any way for the firmware app to know if this time OFF is with effect in the OnOff attribute change callback?

bzbarsky-apple, 2022-05-22:
@jmartinez-silabs Can you please take a look?

jmartinez-silabs, 2022-05-22:
I will take a look

closed 2022-06-13 by andy31415