project-chip/connectedhomeip #15999: [TC-IDM-3.1] and [TC-IDM-3.2]-These Test-Cases fails to Write the value for one attribute on all Endpoints(wildcard)

Rajashreekalmane, 2022-03-09:
#### Problem
As per Test-Plan, [TC-IDM-3.1] and [TC-IDM-3.2]-These Test-Cases step 2 fails to Write the value for one attribute on all  Endpoints(endpoint as wildcard).

 * expected behavior:-DUT should write the value for attribute on all endpoints 
 * actual behavior:-INVALID_ACTION(0x80)

 * steps to reproduce:-sudo ./chip-tool any write-by-id  0x0008 0x0011 1 1 0xffff

Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-3-1-write-request-action-from-dut-to-th-dut-controller

 PFA logs below:-
[TC-IDM-3.1.txt](https://github.com/project-chip/connectedhomeip/files/8212856/TC-IDM-3.1.txt)

bzbarsky-apple, 2022-03-09:
The SDK does not support wildcard writes, and I believe the plan is to not support them for 1.0.

mrjerryjohns, 2022-03-17:
Not to be tested for v1.0.

closed 2022-03-17 by mrjerryjohns