project-chip/connectedhomeip #16018: [TC-MF-1.6] RevokeCommissioning commands seems to always fail when used in YAML scripts

kvikrambhat, 2022-03-09:
RevokeCommissioning commands seems to always fail when used in YAML scripts, the commands seems to work fine when used with the chip-tool CLI.

Attaching screenshots for reference 
1.Test plan screenshot
[image]
2.YAML logs
[revokecommandfail.txt](https://github.com/project-chip/connectedhomeip/files/8216141/revokecommandfail.txt)
[image]

raju-apple, 2022-03-09:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/TestDiscovery.yaml#L290

kvikrambhat, 2022-03-10:
Closing issue , command works fine when "timedInteractionTimeoutMs" argument is used.

closed 2022-03-10 by kvikrambhat