project-chip/connectedhomeip #16037: TE8 Mac Chip Tool Darwin MultiFabric Testing: Opening commissioning window giving 0xc3 error

Garik098, 2022-03-09:
While testing MultiFabric using Chip-tool-darwin, the opening commissioning window either in ECM or in BCM is giving 0xc3 error. Optional parameters supported by Chip-tool might be needed for Chip-tool-darwin to pass the opening commissioning window phase.

Below is the log for opening commissioning window using chip-tool-darwin using ECM
adk@adks-Mac-mini connectedhomeip % ./out/debug/chip-tool-darwin administratorcommissioning open-commissioning-window 500 3840 1000 1 1 1
[1646862247372] [7979:480995] CHIP: [TOO] Running Command
2022-03-09 13:44:07.376 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: ChipToolOpCredsCAKey, value BAG6TxndA16E328KB+s67FX9+5LHU3I7cBrMiHJcajq1An4MxepBioxfnjaunMfI0vAeiWTeGU8HTgmQJVfHOAVnEv1hZWowEttLnZKxu5hy5AIn9FY0u9QOq48M2kXqmA==
2022-03-09 13:44:07.379 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: com.zigbee.chip.commissioner.device_id
2022-03-09 13:44:07.379 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: com.zigbee.chip.commissioner.device_id, value Gr6DOjXiCDA=
2022-03-09 13:44:07.380 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: MatterCARootCert0
2022-03-09 13:44:07.380 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: MatterCARootCert0, value MIIB2TCCAYCgAwIBAgIBADAKBggqhkjOPQQDAjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwHhcNMjIwMzA4MTUwMDAxWhcNMjMwMzA4MTUwMDAxWjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAAQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFo2MwYTAPBgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBBjAdBgNVHQ4EFgQUpAxXUsDrlQkg7iOO2exDAIyjA7wwHwYDVR0jBBgwFoAUpAxXUsDrlQkg7iOO2exDAIyjA7wwCgYIKoZIzj0EAwIDRwAwRAIgQUDQA6aWAftkPOHso2ZV3++c1BRhHsD1MZ3vK0lauJcCIEZEVgTELLz7k+XFXsETqa4U1jXFK3GF+Zyi2DvqqfmU
[1646862247380] [7979:480995] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1646862247400] [7979:480995] CHIP: [DL] _Init
[1646862247400] [7979:480995] CHIP: [IN] UDP::Init bind&listen port=5541
[1646862247401] [7979:480995] CHIP: [IN] UDP::Init bound to port=5541
[1646862247401] [7979:480995] CHIP: [IN] UDP::Init bind&listen port=5541
[1646862247401] [7979:480995] CHIP: [IN] UDP::Init bound to port=5541
[1646862247401] [7979:480995] CHIP: [IN] TransportMgr initialized
[1646862247401] [7979:480995] CHIP: [DIS] Init fabric pairing table with server storage
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F01/Fabric1
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F01/Fabric1, value AQAAAAAAAAAavoM6NeIIMAAAAAAAAAAAAADvAAAA/AAE4DFh8ZQ0tNQWhy2zZhR7hKLLXtBT7c4nqAys8VsPVAmVfY4IWOkwCGvh7S16/4xP1mPfxNJWTSXpsM8kaIYYrwE1VKTVzrgo21ldtxwiPn8R9I81i2JRVt1bLMfrqwYlAAAAAAAAAGEAAAAAAAAAFTABAQAkAgE3AyUU0gQkFQAYJgTxK7opJgVxX5srNwYlFNIEJBUAGCQHASQIATAJQQQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFNwo1ASkBGCQCYDAEFKQMV1LA65UJIO4jjtnsQwCMowO8MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAQUDQA6aWAftkPOHso2ZV3++c1BRhHsD1MZ3vK0lauJdGRFYExCy8+5PlxV7BE6muFNY1xStxhfmcotg76qn5lBgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAVMAEBASQCATcDJRTSBCQVABgmBPEruikmBXFfmys3BiQVACcRGr6DOjXiCDAYJAcBJAgBMAlBBOAxYfGUNLTUFocts2YUe4Siy17QU+3OJ6gMrPFbD1QJlX2OCFjpMAhr4e0tev+MT9Zj38TSVk0l6bDPJGiGGK83CjUBKAEYJAIBNgMEAgQBGDAEFE5MrUA6J9TLEXIDswqDh+XQhN25MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAwaRsIg0BJ16NFDi9Tei62tsnhrhlv/DGr3JxPjTx+MDE1m1JyVObAGfLE4fUlLHz7sNhWEgCBjqEi5ZNR6ZVQRgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
[1646862247402] [7979:480995] CHIP: [IN] Loading certs from storage
[1646862247402] [7979:480995] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000000 and root pubkey
[1646862247402] [7979:480995] CHIP: [IN] 0x04, 0x01, 0xba, 0x4f, 0x19, 0xdd, 0x03, 0x5e, 
[1646862247402] [7979:480995] CHIP: [IN] 0x84, 0xdf, 0x6f, 0x0a, 0x07, 0xeb, 0x3a, 0xec, 
[1646862247402] [7979:480995] CHIP: [IN] 0x55, 0xfd, 0xfb, 0x92, 0xc7, 0x53, 0x72, 0x3b, 
[1646862247402] [7979:480995] CHIP: [IN] 0x70, 0x1a, 0xcc, 0x88, 0x72, 0x5c, 0x6a, 0x3a, 
[1646862247402] [7979:480995] CHIP: [IN] 0xb5, 0x02, 0x7e, 0x0c, 0xc5, 0xea, 0x41, 0x8a, 
[1646862247402] [7979:480995] CHIP: [IN] 0x8c, 0x5f, 0x9e, 0x36, 0xae, 0x9c, 0xc7, 0xc8, 
[1646862247402] [7979:480995] CHIP: [IN] 0xd2, 0xf0, 0x1e, 0x89, 0x64, 0xde, 0x19, 0x4f, 
[1646862247402] [7979:480995] CHIP: [IN] 0x07, 0x4e, 0x09, 0x90, 0x25, 0x57, 0xc7, 0x38, 
[1646862247402] [7979:480995] CHIP: [IN] 0x05, 
[1646862247402] [7979:480995] CHIP: [IN] Generated compressed fabric ID
[1646862247402] [7979:480995] CHIP: [IN] 0xd0, 0x99, 0xe4, 0x0d, 0x93, 0xb1, 0xdc, 0x0c, 
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F02/Fabric2
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F02/Fabric2, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F03/Fabric3
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F03/Fabric3, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F04/Fabric4
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F04/Fabric4, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F05/Fabric5
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F05/Fabric5, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F06/Fabric6
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F06/Fabric6, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F07/Fabric7
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F07/Fabric7, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F08/Fabric8
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F08/Fabric8, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F09/Fabric9
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F09/Fabric9, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F0A/Fabrica
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F0A/Fabrica, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F0B/Fabricb
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F0B/Fabricb, value (null)
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] PersistentStorageDelegate Sync Get Value for Key: F0C/Fabricc
2022-03-09 13:44:07.402 chip-tool-darwin[7979:480995] CHIPPersistentStorageDelegate Get Value for Key: F0C/Fabricc, value (null)
[... cut ...]
[1646864504525] [8093:495874] CHIP: [DMG] IM WH moving to [Uninitialized]
[1646864504525] [8093:495874] CHIP: [BLE] CancelConnection
[1646864504525] [8093:495874] CHIP: [DL] Inet Layer shutdown
[1646864504525] [8093:495874] CHIP: [DL] BLE shutdown
[1646864504525] [8093:495874] CHIP: [DL] System Layer shutdown
[1646864504525] [8093:495869] CHIP: [-] IM Error 0x000005C3: General error: 0xc3 at ../../commands/common/CHIPCommandBridge.mm:51
[1646864504525] [8093:495869] CHIP: [TOO] Run command failure: IM Error 0x000005C3: General error: 0xc3

bzbarsky-apple, 2022-03-11:
Apart from whatever #16050 fixes, `0xC3` is `UNSUPPORTED_CLUSTER`.  That's because the command line above is trying to use the administrator commissioning cluster on endpoint 1, where it does not exist...

@Garik098 what happens if you use the right endpoint (endpoint 0)?

Garik098, 2022-03-11:
@bzbarsky-apple With endpoint 0 I am seeing 0x02 (cluster specific error). On the accessory logs it shows failed to open commissioning window. Below is the controller side log.

adk@adks-Mac-mini debug % ./chip-tool-darwin administratorcommissioning open-commissioning-window 500 3840 1000 1 1 0
[1647023778890] [5075:453188] CHIP: [TOO] Running Command
2022-03-11 10:36:18.895 chip-tool-darwin[5075:453188] CHIPPersistentStorageDelegate Get Value for Key: ChipToolOpCredsCAKey, value BAG6TxndA16E328KB+s67FX9+5LHU3I7cBrMiHJcajq1An4MxepBioxfnjaunMfI0vAeiWTeGU8HTgmQJVfHOAVnEv1hZWowEttLnZKxu5hy5AIn9FY0u9QOq48M2kXqmA==
2022-03-11 10:36:18.897 chip-tool-darwin[5075:453188] PersistentStorageDelegate Sync Get Value for Key: com.zigbee.chip.commissioner.device_id
2022-03-11 10:36:18.897 chip-tool-darwin[5075:453188] CHIPPersistentStorageDelegate Get Value for Key: com.zigbee.chip.commissioner.device_id, value Gr6DOjXiCDA=
2022-03-11 10:36:18.898 chip-tool-darwin[5075:453188] PersistentStorageDelegate Sync Get Value for Key: MatterCARootCert0
2022-03-11 10:36:18.898 chip-tool-darwin[5075:453188] CHIPPersistentStorageDelegate Get Value for Key: MatterCARootCert0, value MIIB2TCCAYCgAwIBAgIBADAKBggqhkjOPQQDAjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwHhcNMjIwMzA4MTUwMDAxWhcNMjMwMzA4MTUwMDAxWjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAAQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFo2MwYTAPBgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBBjAdBgNVHQ4EFgQUpAxXUsDrlQkg7iOO2exDAIyjA7wwHwYDVR0jBBgwFoAUpAxXUsDrlQkg7iOO2exDAIyjA7wwCgYIKoZIzj0EAwIDRwAwRAIgQUDQA6aWAftkPOHso2ZV3++c1BRhHsD1MZ3vK0lauJcCIEZEVgTELLz7k+XFXsETqa4U1jXFK3GF+Zyi2DvqqfmU
[1647023778898] [5075:453188] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1647023778905] [5075:453188] CHIP: [DL] _Init
[1647023778905] [5075:453188] CHIP: [IN] UDP::Init bind&listen port=5541
[1647023778907] [5075:453188] CHIP: [IN] UDP::Init bound to port=5541
[1647023778907] [5075:453188] CHIP: [IN] UDP::Init bind&listen port=5541
[1647023778907] [5075:453188] CHIP: [IN] UDP::Init bound to port=5541
[1647023778907] [5075:453188] CHIP: [IN] TransportMgr initialized
[1647023778907] [5075:453188] CHIP: [DIS] Init fabric pairing table with server storage
2022-03-11 10:36:18.907 chip-tool-darwin[5075:453188] PersistentStorageDelegate Sync Get Value for Key: F01/Fabric1
2022-03-11 10:36:18.907 chip-tool-darwin[5075:453188] CHIPPersistentStorageDelegate Get Value for Key: F01/Fabric1, value AQAAAAAAAAAavoM6NeIIMAAAAAAAAAAAAADvAAAA/AAE4DFh8ZQ0tNQWhy2zZhR7hKLLXtBT7c4nqAys8VsPVAmVfY4IWOkwCGvh7S16/4xP1mPfxNJWTSXpsM8kaIYYrwE1VKTVzrgo21ldtxwiPn8R9I81i2JRVt1bLMfrqwYlAAAAAAAAAGEAAAAAAAAAFTABAQAkAgE3AyUU0gQkFQAYJgTxK7opJgVxX5srNwYlFNIEJBUAGCQHASQIATAJQQQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFNwo1ASkBGCQCYDAEFKQMV1LA65UJIO4jjtnsQwCMowO8MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAQUDQA6aWAftkPOHso2ZV3++c1BRhHsD1MZ3vK0lauJdGRFYExCy8+5PlxV7BE6muFNY1xStxhfmcotg76qn5lBgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAVMAEBASQCATcDJRTSBCQVABgmBPEruikmBXFfmys3BiQVACcRGr6DOjXiCDAYJAcBJAgBMAlBBOAxYfGUNLTUFocts2YUe4Siy17QU+3OJ6gMrPFbD1QJlX2OCFjpMAhr4e0tev+MT9Zj38TSVk0l6bDPJGiGGK83CjUBKAEYJAIBNgMEAgQBGDAEFE5MrUA6J9TLEXIDswqDh+XQhN25MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAwaRsIg0BJ16NFDi9Tei62tsnhrhlv/DGr3JxPjTx+MDE1m1JyVObAGfLE4fUlLHz7sNhWEgCBjqEi5ZNR6ZVQRgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
[1647023778907] [5075:453188] CHIP: [IN] Loading certs from storage
[1647023778907] [5075:453188] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000000 and root pubkey
[1647023778907] [5075:453188] CHIP: [IN] 0x04, 0x01, 0xba, 0x4f, 0x19, 0xdd, 0x03, 0x5e, 
[1647023778907] [5075:453188] CHIP: [IN] 0x84, 0xdf, 0x6f, 0x0a, 0x07, 0xeb, 0x3a, 0xec, 
[1647023778907] [5075:453188] CHIP: [IN] 0x55, 0xfd, 0xfb, 0x92, 0xc7, 0x53, 0x72, 0x3b, 
[1647023778907] [5075:453188] CHIP: [IN] 0x70, 0x1a, 0xcc, 0x88, 0x72, 0x5c, 0x6a, 0x3a, 
[1647023778907] [5075:453188] CHIP: [IN] 0xb5, 0x02, 0x7e, 0x0c, 0xc5, 0xea, 0x41, 0x8a, 
[1647023778907] [5075:453188] CHIP: [IN] 0x8c, 0x5f, 0x9e, 0x36, 0xae, 0x9c, 0xc7, 0xc8, 
[1647023778907] [5075:453188] CHIP: [IN] 0xd2, 0xf0, 0x1e, 0x89, 0x64, 0xde, 0x19, 0x4f, 
[1647023778907] [5075:453188] CHIP: [IN] 0x07, 0x4e, 0x09, 0x90, 0x25, 0x57, 0xc7, 0x38, 
[1647023778907] [5075:453188] CHIP: [IN] 0x05, 
[... cut ...]
[1647023778958] [5075:453192] CHIP: [DMG] IM WH moving to [Uninitialized]
[1647023778958] [5075:453192] CHIP: [BLE] CancelConnection
[1647023778958] [5075:453192] CHIP: [DL] Inet Layer shutdown
[1647023778958] [5075:453192] CHIP: [DL] BLE shutdown
[1647023778958] [5075:453192] CHIP: [DL] System Layer shutdown
[1647023778958] [5075:453188] CHIP: [-] IM Error 0x00000602: Cluster-specific error: 0x02 at ../../commands/common/CHIPCommandBridge.mm:51
[1647023778958] [5075:453188] CHIP: [TOO] Run command failure: IM Error 0x00000602: Cluster-specific error: 0x02

krypton36, 2022-03-23:
@Garik098 I was able to verify opening commissioning window and pairing. I used the following command and manual code:
`chip-tool-darwin administratorcommissioning open-commissioning-window 500 hex:06c756dffcd722653452a12dcd945d8c54da2b0f3cbd1b4dc3f1adb223aeb26b047cd24c96866f979b1d83ec50e2b4ae30cdf2fdb32bd8a211b837dc94edcd56f4d14377191076bfc59d99b7dd3053efd6f02c4434f2bdd27aa4f9cea70d738e4c 3840 1000 hex:5350414b453250204b65792053616c74 1 0`

manual pairing code: `36200325747`

Garik098, 2022-04-04:
With the manual pairing code 34969029007 and the above open commissioning window command, I was able to do the ECM pairing scenario.