project-chip/connectedhomeip #16053: Commissioning fails when setting Discriminator to 3842 - CHIP Error 0x0000002F: Invalid argument

kean-apple, 2022-03-10:
#### Problem
Commissioning fails when setting Discriminator to 3842 - CHIP Error 0x0000002F: Invalid argument

SHA: 4279b678c16453b332d8e3108098b678a2940df6

1. Modify discriminator value to 0xF02 in https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CHIPDeviceConfig.h#L900

#define CHIP_DEVICE_CONFIG_USE_TEST_SETUP_DISCRIMINATOR 0xF02

2. Build M5 board with new discriminator value
3. Try to pair with iOS chiptool - it fails with: 
2022-03-09 19:39:28.998384-0800  localhost CHIPTool[9143]: (CHIP) DevicePairingDelegate Commissioning complete. NodeId 1 Status ../../../../../../../../../../../connectedhomeip/src/controller/AutoCommissioner.cpp:159: CHIP Error 0x0000002F: Invalid argument

In the past I was able to pair with different discriminator values other than default. The default 0xF00 seems to work
[pairing-fail-change-setup-discriminator-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8220230/pairing-fail-change-setup-discriminator-chiptool-logs.txt)
[commisioning-fails-change-setup-discriminator.txt](https://github.com/project-chip/connectedhomeip/files/8220232/commisioning-fails-change-setup-discriminator.txt)

kean-apple, 2022-03-10:
Added another set of logs using default discriminator
[failed-to-pair-invalid-argument-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8224528/failed-to-pair-invalid-argument-M5-logs.txt)
[failed-pairing-invalid-argument-chiptool-log.txt](https://github.com/project-chip/connectedhomeip/files/8224529/failed-pairing-invalid-argument-chiptool-log.txt)

bzbarsky-apple, 2022-03-10:
> Build M5 board with iOS chiptool

I assume that's with all-clusters-app?

kean-apple, 2022-03-10:
Yes all clusters app

cecille, 2022-03-10:
The PairDevice call for setup codes where the device is discovered first over BLE was inconsistent with the rest of the PairDevice calls such that if you discovered a device over BLE you would establish a pase session only, whereas if you discovered it over mdns first you would get a full commissioning. This was a race condition on the behaviour.

If you want to just establish PASE, there was a new call added that will do that - EstablishPASEConnection with a setup code. 

Does this happen in the iOS chip tool? If so, I can prepare a patch to update, but I don't have a good way to test this as it's not in the CI and I don't have a way to test.

bzbarsky-apple, 2022-03-10:
@cecille So I am at a loss as to how this is supposed to work now.  The way it used to work was:
1. We would call PairDevice with a code.
2. If the device was discovered on-network, it would get commissioned.
3. If the device was discovered via BLE we would establish PASE and call back into the "pairing complete" bits to allow the consumer to get network info and call Commission.

In the new setup, if I call PairDevice that will try to commission and fail out.

If I call EstablishPASEConnection, that works for the BLE case, but now the on-network case is broken: we get the "pairing complete" callback, but there is no way to tell that network info does not need to be provided in this case, so we end up trying to provide it, which is not the desired flow here.

Given the current API surface, how do we get the behavior that we want, exactly? 

> Does this happen in the iOS chip tool?

Yes, exactly.

closed 2022-03-10 by bzbarsky-apple