project-chip/connectedhomeip #16069: Matter Bridge application on linux platform crashes on a multi fabrics scenario

Carpao79, 2022-03-10:
#### Problem
By testing Multi Fabrics scenarios on a bridge application running on linux platform, sometimes the application crashes, mainly while commissioning the second or following fabric.
Here the DUT logs of third fabric commissioning:
> [1646918447.219150][19054:19056] CHIP:IN: Generated compressed fabric ID
[1646918447.220386][19054:19056] CHIP:IN: 0xd9, 0x98, 0x61, 0xbf, 0x0d, 0x44, 0x30, 0x27,
[1646918447.221595][19054:19056] CHIP:DIS: Added new fabric at index: 3, Initialized: 1
[1646918447.221776][19054:19056] CHIP:DIS: Assigned compressed fabric ID: 0xD99861BF0D443027, node ID: 0x0000000000000001
[1646918447.221895][19054:19056] CHIP:DIS: Fabric (3) loaded from storage. Calling OnFabricRetrievedFromStorage
[1646918447.222439][19054:19056] CHIP:ZCL: OpCreds: Fabric 0x3 was retrieved from storage. FabricId 0x0000000000000001, NodeId 0x0000000000000001, VendorId 0xFFF1
[1646918447.226475][19054:19056] CHIP:ZCL: OpCreds: Call to fabricListChanged
[1646918447.226764][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faec9
[1646918447.227825][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faeca
[1646918447.236836][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-cAgfOB)
[1646918447.696430][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918447.696737][19054:19056] CHIP:DIS: Fabric (3) persisted to storage. Calling OnFabricPersistedToStorage
[1646918447.696836][19054:19056] CHIP:ZCL: OpCreds: Fabric 3 was persisted to storage. FabricId 0000000000000001, NodeId 0000000000000001, VendorId 0xFFF1
[1646918447.696924][19054:19056] CHIP:ZCL: OpCreds: Call to fabricListChanged
[1646918447.697012][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faecb
[1646918447.702171][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faecc
[1646918447.704073][19054:19056] CHIP:DIS: Fabric (3) loaded from storage. Calling OnFabricRetrievedFromStorage
[1646918447.705079][19054:19056] CHIP:ZCL: OpCreds: Fabric 0x3 was retrieved from storage. FabricId 0x0000000000000001, NodeId 0x0000000000000001, VendorId 0xFFF1
[1646918447.705366][19054:19056] CHIP:ZCL: OpCreds: Call to fabricListChanged
[1646918447.705488][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faecd
[1646918447.705632][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faece
[1646918447.714348][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-pfKfMI)
[1646918448.423878][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918448.425652][19054:19056] CHIP:DIS: Fabric (3) persisted to storage. Calling OnFabricPersistedToStorage
[1646918448.430355][19054:19056] CHIP:ZCL: OpCreds: Fabric 3 was persisted to storage. FabricId 0000000000000001, NodeId 0000000000000001, VendorId 0xFFF1
[1646918448.431884][19054:19056] CHIP:ZCL: OpCreds: Call to fabricListChanged
[1646918448.433239][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faecf
[1646918448.434559][19054:19056] CHIP:DMG: Endpoint 0, Cluster 0x0000_003E update version to 421faed0
[1646918448.436606][19054:19056] CHIP:DMG: AccessControl: validating f=3 p=a a=c s=1 t=0
[1646918448.441113][19054:19056] CHIP:DMG:   validating subject 0x000000000001B669
[1646918448.444082][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-mcOJlJ)
[1646918448.783148][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918448.786502][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-NEGpEb)
[1646918449.128265][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918449.131887][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-kjECpp)
[1646918449.473976][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918449.477551][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-jHpHFG)
[1646918449.863188][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918449.867510][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-laBhLP)
[1646918450.208209][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918450.211240][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-KpClgJ)
[1646918450.556488][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918450.565699][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-fofIiO)
[1646918451.017164][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918451.024265][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-BoEoND)
[1646918451.537045][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918451.539935][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-OBolDG)
[1646918451.886575][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918451.889886][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-lnpnKl)
[1646918452.234646][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918452.237901][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-LAAMCF)
[1646918452.577453][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918452.580733][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-CLpPMd)
[1646918452.922702][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918452.926036][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-HPEDAJ)
[1646918453.422656][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918453.425919][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-dLMhHi)
[1646918453.773216][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918453.776236][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-EdGjjP)
[1646918454.401218][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918454.404469][19054:19056] CHIP:DL: writing settings to file (/etc/chip_kvs-pfOPIL)
[1646918454.784209][19054:19056] CHIP:DL: renamed tmp file to file (/etc/chip_kvs)
[1646918454.785518][19054:19056] CHIP:ZCL: OpCreds: ACL entry created for Fabric 3 CASE Admin NodeId 0x000000000001B669
[1646918454.788895][19054:19056] CHIP:DL: Using wifi MAC for hostname
[1646918454.790048][19054:19056] CHIP:DIS: Advertise operational node C8A8379416C1044D-0000000000000001
[1646918454.795335][19054:19056] CHIP:DIS: CHIP minimal mDNS configured as 'Operational device'.
[1646918454.810765][19054:19056] CHIP:DIS: Broadcasting mDns reply for query from fe80::c693:ff:fe12:abba
Segmentation fault

I tried the same against the `lighting-app` example application that comes with the TE8 raspberry image but the crash seems not reproducible. Anyway after the commissioning of two fabrics the `avahi-browse` command shows the following:
> ubuntu@ubuntu:~/matter/chip-tool-tmp$ avahi-browse -rt _matter._tcp
+   eth0 IPv6 D99861BF0D443027-0000000000000001             _matter._tcp         local
+   eth0 IPv6 C8A8379416C1044D-0000000000000001             _matter._tcp         local
=   eth0 IPv6 D99861BF0D443027-0000000000000001             _matter._tcp         local
   hostname = [E45F011B40CA0000.local]
   address = [fe80::e65f:1ff:fe1b:40ca]
   port = [5540]
   txt = ["T=1" "▒Q▒ê▒"]
=   eth0 IPv6 C8A8379416C1044D-0000000000000001             _matter._tcp         local
   hostname = [E45F011B40CA0000.local]
   address = [fe80::e65f:1ff:fe1b:40ca]
   port = [5540]
   txt = ["T=1" "CRA=300" "CRI=5000"]

One of the TXT record seems malformed.
Maybe the two issues have the same root cause.

bzbarsky-apple, 2022-06-13:
@Carpao79 Is this still happening?  Can you run under a debugger and get a stack?

Carpao79, 2022-06-14:
> @Carpao79 Is this still happening? Can you run under a debugger and get a stack?

@bzbarsky-apple it looks like the issue is not reproducible anymore with SDK coming from the TE9 SHA.

bzbarsky-apple, 2022-06-14:
Thank you for double-checking!

closed 2022-06-14 by bzbarsky-apple