project-chip/connectedhomeip #16096: [TC-DD-3.1] Need to print different stages of pairing in the log file as we cant verify all the commissiong flow in the current lof file

sowmyassp, 2022-03-11:
#### Problem
Need to print those different stages of pairing in the log file
 
* expected behavior 
     To get all the commissioning flow in the log file after commissioning DUT with TH 
 * actual behavior
     Some of commissioning flow is seen in the current log
 * steps to reproduce
     DUT - sudo ./chip-all-clusters-app --wifi
     TH - ./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840
 * screenshots, images, logs, etc.
    [BLE_provisioning_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8229377/BLE_provisioning_cntrl.txt)
    [BLE_provisiong_DUT.txt](https://github.com/project-chip/connectedhomeip/files/8229378/BLE_provisiong_DUT.txt)

bzbarsky-apple, 2022-03-11:
@sowmyassp does https://github.com/project-chip/connectedhomeip/pull/14736 address this?

sowmyassp, 2022-03-31:
@bzbarsky-apple  yes, this ticket addresses the issue

bzbarsky-apple, 2022-04-01:
Please close if resolved?

sowmyassp, 2022-04-05:
No its not yet resolved, for these [TC-DD-3.1],  [TC-DD-3.2], [TC-DD-3.5], [TC-DD-3.6] test-cases we need commissioning flow such as "Commissioner configures Fabric information", "Commissioner requests operational CSR" which is needed to verify this test-cases.

andy31415, 2022-04-19:
@sowmyassp - could you describe the reason and use of this printout (likely subset of the actual test case description)?

Release builds of devices (such as devices being certified) would not get access to logs so having tests rely on logs and their content does not seem the best way, however I am not convinced we have an ideal way either. Is there a requirement for certified devices to have logging enabled? How about tiny devices that cannot even have logging (e.g. qpg devices had logging disabled to use less flash).

cjandhyala, 2022-04-20:
@andy31415 The log lines requested are on the TH side but not DUT. TCs like https://github.com/CHIP-Specifications/chip-test-plans/blob/3bcd71e5bf2fe7ce7b143614f9b0ee89c9af2b16/src/devicediscovery.adoc#335-tc-dd-35-commissioning-flow---concurrent-dut---commissioner , require to verify every step of the commissioning flow. The only we can do that is by verifying the log lines in the TH to see if DUT finished all those intermediate steps in the commissioning flow.  Alternatively we need to think, do we need to verify all those intermediate steps of commissioning flow as part of certification test case OR just verify if the device can be provisioned successfully or not ?

sowmyassp, 2022-05-13:
Test-plan is updated with correct commissioning flow and we are getting the same in logs

sowmyassp, 2022-05-13:
PFA Logs: 
[commissioning_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8686214/commissioning_cntrl.txt)
[commissioning_dut.txt](https://github.com/project-chip/connectedhomeip/files/8686215/commissioning_dut.txt)

closed 2022-05-13 by sowmyassp