project-chip/connectedhomeip #16099: [TC-NC-1.2] LastConnectErrorValue attribute value should be null, but is 0 [TE8]

markus-becker-tridonic-com, 2022-03-11:
#### Problem

3.2. [TC-NC-1.2] Verification on thread connection and attributes check [DUT - Commissionee

Step 12
Expected Outcome: Verify that LastConnectErrorValue attribute value is null

```
[1646848244.189667][26383:26388] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0007DataVersion: 4042607948
[1646848244.189799][26383:26388] CHIP:TOO:   LastConnectErrorValue: 0
```

null != 0 or?

Spec:
7 LastConnectErrorValue int32 X null R A M

 #### Proposed Solution

Return null by default.

bzbarsky-apple, 2022-03-11:
@yufengwangca

bzbarsky-apple, 2022-06-18:
@markus-becker-tridonic-com Is this still happening on tip?

woody-apple, 2022-06-21:
Cert Blocker Review: Given t his is feedback from an older TE, closing. Please re-open if this is still happening.