project-chip/connectedhomeip #16100: [TC-IDM-6.4]-While verifying the Activation of the Events Subscribe Interaction, Chip-Tool aborted with a "Segmentation fault" during shutdown.

Rajashreekalmane, 2022-03-11:
#### Problem
While verifying the Activation of the Events Subscribe Interaction, Chip-Tool aborted with "Segmentation fault" and Run command failure during shutdown.

Setup:
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:-5e4a98cbb9362eee073e6e58e9ea0d6032112146					

 * steps to reproduce:-
 sudo ./chip-tool any subscribe-event-by-id 0x0033 0x01 0 100 0 1 1
sudo ./chip-tool any subscribe-event-by-id 0xFFFFFFFF 0xFFFFFFFF 0 100 0 1 0

PFA Screenshot:
[image]

PFA complete log below:-
[Event subscription.txt](https://github.com/project-chip/connectedhomeip/files/8230133/Event.subscription.txt)

Rajashreekalmane, 2022-04-05:
Still facing the same issue for the latest commit 41a431dcc7c123699818636afd01b57412a6f3a4

PFA log
[IDM -event segmentation.txt](https://github.com/project-chip/connectedhomeip/files/8417174/IDM.-event.segmentation.txt)

vivien-apple, 2022-04-29:
Can you try to run `chip-tool interactive start` and when in interactive mode do:
```
any subscribe-event-by-id 0x0033 0x01 0 100 1 1
```

For the fact that the command were not closing `chip-tool` properly it was because of https://github.com/project-chip/connectedhomeip/pull/17842