project-chip/connectedhomeip #16102: [TC-IDM-5.2]-Chip-tool need to have a provision to write functionality for unsuported attributes.

Rajashreekalmane, 2022-03-11:
#### Problem
To verify the DUT whether it is written unsupported access Message. The Chip-tool need to support write Access for unsupported attributes.

Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-5-2-timed-request-action-th-to-dut

bzbarsky-apple, 2022-03-11:
@Rajashreekalmane Is this talking about YAML or manual chip-tool invocation?

Does "unsupported" mean "unsupported on the server but defined in the spec", or does "unsupported" mean "not defined in the spec at all"?

bzbarsky-apple, 2022-03-11:
And @Rajashreekalmane what does the timed request test plan have to do with this issue?

sankarselvam, 2022-03-17:
@Rajashreekalmane  This is not a valid issue. You are looking for a Chip-tool support to send a delayed write request to verify the Test case TC-IDM-5.2. 
Please close this issue.

closed 2022-03-18 by Rajashreekalmane