project-chip/connectedhomeip #16108: [SQA] [TC-OO-2.3] - Step 10e doesn't return expected outcome

Survensa, 2022-03-11:
[TC-OO-2.3] 
In the above mentioned test-case in step 10e we are facing issue as explained below
Expected outcome - on-off value should be true
Actual Outcome -  on-off value  should be false
Steps to reproduce - ./chip-tool onoff read on-off 949 1
Attached Logs:
[tc-oo-2.3 error log.txt](https://github.com/project-chip/connectedhomeip/files/8231429/tc-oo-2.3.error.log.txt)

bzbarsky-apple, 2022-03-11:
@jmartinez-silabs could you take a look?

jmartinez-silabs, 2022-03-11:
Testing it on my end. Work correctly for me.

First parameter of the command is set to 1, which means :
`The AcceptOnlyWhenOn sub-field is 1 bit in length and specifies whether the OnWithTimedOff command is to be processed unconditionally or only when the OnOff attribute is equal to TRUE. If this sub-field is set to 1, the OnWithTimedOff command SHALL only be accepted if the OnOff attribute is equal to TRUE. If this sub-field is set to 0, the OnWithTimedOff command SHALL be processed unconditionally.`

I don't see in your log if the onoff attribute was on (TRUE) before starting the test but you need it else all commands are ignored in this case.

I will let you test it out again before closing the ticket

Survensa, 2022-03-14:
I see it works fine with the latest commit id 2982c0ca0623773327def4cc7180d07f6b959c42

closed 2022-03-14 by Survensa