project-chip/connectedhomeip #16114: [TC-DM-1.2], [TC-DM-1.4] -  Events can't be verified

sowmyassp, 2022-03-11:
#### Problem
Events can't be verified - shut-down, leave event 
 * expected behavior : after the execution of events no more events or commands or attributes can't be executed
 * actual behavior : after the execution of events other commands can be executed 
 * steps to reproduce : sudo ./chip-tool basic read-event start-up 1 0
                                    sudo ./chip-tool basic read-event shut-down 1 0
                                    sudo ./chip-tool basic read-event leave 1 0

bzbarsky-apple, 2022-03-11:
I don't understand what this issue is about, exactly...  read-event for the "startup" event will generally work and other things can happen after that, of course.

Reading the shutdown and leave events will generally return nothing, because those only happen when the server is shutting down or being decomissioned, so is no longer around to be read from.  Subscribing to them should work, though.

@sowmyassp can you please describe the exact behavior you expected and why you expected it?

sowmyassp, 2022-03-17:
@bzbarsky-apple ----Yes reading the shutdown and leave events will generally return nothing but after that if we read any other basic attribute or send any command it should not pass as after shutdown event DUT will be closed right?

bzbarsky-apple, 2022-03-18:
If the device is shut down, yes...

At this point, this issue could benefit from:
1. Clear steps to reproduce.  The steps listed above do not involve shutting down the DUT, so I have no idea what they have to do with "after shutdown" behavior.
2. A clear description of where (after which step) expected behavior and actual behavior diverge.

sowmyassp, 2022-03-30:
@bzbarsky-apple  After shutdown event i.e., sudo ./chip-tool basic read-event shut-down 1 0
the connectivity should close and if we try to read for example ./chip-tool basic read vendor-name 1 0  it should fail as we did a shut-down event before.

1. sudo ./chip-tool basic read-event shut-down 1 0
2. After event I tried to read vendor-name
3. sudo ./chip-tool basic read vendor-name 1 0
4. After this command it should fail because in step 1 the window should be closed

bzbarsky-apple, 2022-03-30:
> After shutdown event i.e., sudo ./chip-tool basic read-event shut-down 1 0

Reading the shut-down event does not make anything shut down.

> the connectivity should close

No, it should not.

closed 2022-03-30 by bzbarsky-apple