project-chip/connectedhomeip #16116: [TE8][TC_WNCV_3_1, TC_WNCV_3_2, TC_WNCV_3_4, TC_WNCV_4_3]

eve-cxrp, 2022-03-11:
#### Problem
Chip-Tool tries to read CurrentPositionLiftPercentage although it is marked as optional according to the spec. Our device does not have this attribute. 

 #### Proposed Solution
Check should not fail if it is not present.

bzbarsky-apple, 2022-03-11:
> although it is marked as optional according to the spec

Those reads are gated on PICS bits, as far as I can tell.  Was that PICS set correctly for the test?

jmeg-sfy, 2022-03-11:
Indeed if you can use PICS info the file to filter out those

eve-cxrp, 2022-03-14:
Well, the Test Harness has no selection for PICS? Or did I miss that?

jmeg-sfy, 2022-03-14:
file  /tmp/PICS.txt contains let say:
A_CONFIGSTATUS=0

./out/debug/standalone/chip-tool tests Test_TC_WNCV_2_2 0x12345 --PICS /tmp/PICS.txt

the if = 0 it will skip the test

sadly YAML  cannot simply read the feature map to apply a test (yet hopefully)

then you got
[1643286226417] [32173:84101267] CHIP: [TOO]  ***** Test Step 2 : Reads ConfigStatus attribute from DUT
[1643286226417] [32173:84101267] CHIP: [TOO]  **** Skipping: A_CONFIGSTATUS == false

closed 2022-10-09 by stale[bot]