project-chip/connectedhomeip #16118: [TE8][TC-IDM-2.2] step 5 fails because wildcard reads can easily take longer than 10 seconds

bauerschwan, 2022-03-11:
When using wildcard reads with chip-tool then the internal 10 second timeout for such a read command is sometimes not enough because such reads can return many, many objects depending on the device.
The timeout behaviour should be handled differently and - in case of tools like chip-tool - should be changeable by a command line option.

bauerschwan, 2022-03-11:
Also note that the verification steps for this step is wrong. It should be 
`chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF
`to match the test case description

bzbarsky-apple, 2022-03-11:
Please file an issue in https://github.com/CHIP-Specifications/chip-test-plans on the test plan verification step being wrong?

bauerschwan, 2022-03-11:
Done: https://github.com/CHIP-Specifications/chip-test-plans/issues/938

mrjerryjohns, 2022-05-26:
Sounds like this was completed when the support was added to chip-tool to specify a timeout.