project-chip/connectedhomeip #16124: Commissioning with python chip-device-ctrl fails due to network information not provided

caubut-charter, 2022-03-11:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

dtodor, 2022-03-14:
Not sure about the Python controller, but we are using a custom `CommissioningDelegate` and the same error occurs after the latest changes. The fix is pretty simple inside the `CHIPDeviceController.cpp`:
```
CHIP_ERROR DeviceCommissioner::Commission(NodeId remoteDeviceId, CommissioningParameters & params)
{
    ReturnErrorOnFailure(mAutoCommissioner.SetCommissioningParameters(params));
    return Commission(remoteDeviceId);
}
```
i.e. replace `mAutoCommissioner.` with `mDefaultCommissioner->`.

caubut-charter, 2022-04-06:
No longer having this issue.

closed 2022-04-06 by caubut-charter