project-chip/connectedhomeip #16129: [TC-MC-9.2] Simulated test crashing when made a run

manjunathambiger23, 2022-03-11:
Summary:
               when trying out simulated device test case for "[TC-MC-9.2]" we are observing a crash showing response as : "Response Failure: IM Error 0x000005C3: General error: 0xc3"

Execution logs :
[TC_MC_9_2_app_side.txt](https://github.com/project-chip/connectedhomeip/files/8233998/TC_MC_9_2_app_side.txt)
[TC_MC_9_2_CHip_tool.txt](https://github.com/project-chip/connectedhomeip/files/8233999/TC_MC_9_2_CHip_tool.txt)

krypton36, 2022-03-11:
This issue appears to be happening, because `MatterPostAttributeReadCallback` gets called before sending report in `BuildAndSendSingleReportData` which causes the thread to End Early and terminate the message transmission to the controller. The simulated app will need to wait until after the SenReport is complete and then move on to the next test. This will require SDK changes, assigning to @vivien-apple to implement changes.

Steps to reproduce:
1. Generate Attached simplified YAML for simulated app
2. Compile simulated app
3. Start simulated app
4. Pair simulated app with chip-tool
5. Run the following chip-tool command: `chip-tool applicationbasic read application-version 1 1 `

[Test_TC_MC_9_2_Simulated.txt](https://github.com/project-chip/connectedhomeip/files/8235212/Test_TC_MC_9_2_Simulated.txt)

manjunath-grl, 2023-01-05:
This issue needs to be closed.

closed 2023-08-11 by stale[bot]