project-chip/connectedhomeip #16137: [TE8][TC-LC-1.1] Fixed Label cluster

jmeg-sfy, 2022-03-11:
#### Problem
The tests state chip-tool must write into the fixedlabel cluster
BUT there is no write command possible list
Test wants to check the UNSUPPORTED_WRITE

 ./chip-tool fixedlabel write label-list [room:bedroom]

  ./chip-tool fixedlabel command_name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Commands:                                                                           |
  +-------------------------------------------------------------------------------------+
  | * command-by-id                                                                     |
  | * read-by-id                                                                        |
  | * read                                                                              |
  | * write-by-id                                                                       |
  | * subscribe-by-id                                                                   |
  | * subscribe                                                                         |
  | * read-event-by-id                                                                  |
  | * subscribe-event-by-id                                                             |
  +-------------------------------------------------------------------------------------+

ReneJosefsen, 2022-03-17:
Agree, the test-steps is not valid, they should be changed to:

./chip-tool fixedlabel write-by-id 0 [room:bedroom] <nodeId> 0

This seems to return the error 0x88, which is UNSUPPORTED_WRITE

bzbarsky-apple, 2022-05-22:
Please file test plan issues at https://github.com/CHIP-Specifications/chip-test-plans/issues/new/choose

closed 2022-05-22 by bzbarsky-apple