project-chip/connectedhomeip #16148: TE8 Mac Chip Tool Darwin: Responses to commands (that return a value aren't parsed out, but instead seem to be a memory address?

dwgaul, 2022-03-11:
Ran the following command to trigger an identify routine:

`./chip-tool-darwin identify identify 30 0x000000000001B674 1 `

During this 30 second routine, ran the following which generates a response with value from the server:

`./chip-tool-darwin identify identify-query 0x000000000001B674 1`

chip-tool-darwin receives the response but does not seem to parse out the value:

<details> <summary> Response Received </summary>

```
Commands.h:27571: Success
[1646967788859] [57026:1134822] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:369019 on exchange 41230i
[1646967788859] [57026:1134822] CHIP: [EM] Found matching exchange: 41230i, Delegate: 0x7fa15bd04080
[1646967788859] [57026:1134822] CHIP: [EM] Rxd Ack; Removing MessageCounter:1623392 from Retrans Table on exchange 41230i
[1646967788859] [57026:1134822] CHIP: [EM] Removed CHIP MessageCounter:1623392 from RetransTable on exchange 41230i
[1646967788859] [57026:1134822] CHIP: [DMG] ICR moving to [ResponseRe]
[1646967788859] [57026:1134822] CHIP: [DMG] InvokeResponseMessage =
[1646967788859] [57026:1134822] CHIP: [DMG] {
[1646967788859] [57026:1134822] CHIP: [DMG] 	suppressResponse = false, 
[1646967788859] [57026:1134822] CHIP: [DMG] 	InvokeResponseIBs =
[1646967788859] [57026:1134822] CHIP: [DMG] 	[
[1646967788859] [57026:1134822] CHIP: [DMG] 		InvokeResponseIB =
[1646967788859] [57026:1134822] CHIP: [DMG] 		{
[1646967788859] [57026:1134822] CHIP: [DMG] 			CommandDataIB =
[1646967788859] [57026:1134822] CHIP: [DMG] 			{
[1646967788859] [57026:1134822] CHIP: [DMG] 				CommandPathIB =
[1646967788859] [57026:1134822] CHIP: [DMG] 				{
[1646967788859] [57026:1134822] CHIP: [DMG] 					EndpointId = 0x1,
[1646967788859] [57026:1134822] CHIP: [DMG] 					ClusterId = 0x3,
[1646967788859] [57026:1134822] CHIP: [DMG] 					CommandId = 0x0,
[1646967788859] [57026:1134822] CHIP: [DMG] 				},
[1646967788859] [57026:1134822] CHIP: [DMG] 				
[1646967788859] [57026:1134822] CHIP: [DMG] 				CommandData = 
[1646967788859] [57026:1134822] CHIP: [DMG] 				{
[1646967788859] [57026:1134822] CHIP: [DMG] 					0x0 = 26, 
[1646967788859] [57026:1134822] CHIP: [DMG] 				},
[1646967788859] [57026:1134822] CHIP: [DMG] 			},
[1646967788859] [57026:1134822] CHIP: [DMG] 			
[1646967788859] [57026:1134822] CHIP: [DMG] 		},
[1646967788859] [57026:1134822] CHIP: [DMG] 		
[1646967788859] [57026:1134822] CHIP: [DMG] 	],
[1646967788859] [57026:1134822] CHIP: [DMG] 	
[1646967788859] [57026:1134822] CHIP: [DMG] 	InteractionModelRevision = 1
[1646967788859] [57026:1134822] CHIP: [DMG] },
[1646967788859] [57026:1134822] CHIP: [DMG] Received Command Response Data, Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0000
[1646967788859] [57026:1134822] CHIP: [DMG] ICR moving to [AwaitingDe]
[1646967788859] [57026:1134822] CHIP: [EM] Sending Standalone Ack for MessageCounter:369019 on exchange 41230i
[1646967788859] [57026:1134822] CHIP: [IN] Prepared secure message 0x70000949d570 to 0x000000000001B674 (1)  of type 0x10 and protocolId (0, 0) on exchange 41230i with MessageCounter:1623393.
[1646967788859] [57026:1134822] CHIP: [IN] Sending encrypted msg 0x70000949d570 with MessageCounter:1623393 to 0x000000000001B674 (1) at monotonic time: 531103853 msec
[1646967788859] [57026:1134823] CHIP: [CTL] Response Time: PFvPvRKN4chip3app8Clusters8Identify8Commands21IdentifyQueryResponse13DecodableTypeEE+868200798 0.005640 seconds
[1646967788859] [57026:1134822] CHIP: [EM] Flushed pending ack for MessageCounter:369019 on exchange 41230i
2022-03-10 19:03:08.859 chip-tool-darwin[57026:1134823] Values: <CHIPIdentifyClusterIdentifyQueryResponseParams: 0x600001760010>
[1646967788859] [57026:1134823] CHIP: [TOO] Error: ../../third_party/connectedhomeip/src/darwin/Framework/CHIP/CHIPError.mm:212: Success
```

</details>

I did not observe this behavior with the regular chip-tool:

```
[1646967183119] [56860:1130968] CHIP: [DMG] Received Command Response Data, Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0000
[1646967183119] [56860:1130968] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0003 Command 0x0000_0000
[1646967183119] [56860:1130968] CHIP: [TOO]   IdentifyQueryResponse: {
[1646967183119] [56860:1130968] CHIP: [TOO]     timeout: 7
[1646967183119] [56860:1130968] CHIP: [TOO]    }
```

bzbarsky-apple, 2022-05-22:
We are logging the result struct, but that does not actually log it very nicely...

@jtung-apple @krypton36 is there some "automatic" way to have it log all the struct members and their values?

krypton36, 2022-05-24:
> We are logging the result struct, but that does not actually log it very nicely...
> 
> @jtung-apple @krypton36 is there some "automatic" way to have it log all the struct members and their values?

@bzbarsky-apple We can add a description function to the class in the ZAP generation with it printing the values.

bzbarsky-apple, 2022-05-26:
Note that IdentifyQuery is gone, so this issue will need different steps to reproduce.

PR #18909 (2022-06-02): Add chip tool Darwin description to classes.
#### Problem
When logging a structure class, it does not log any of the attributes. This makes it difficult to validate responses using darwin-framework-tool.
- Fixes #18634 
- Fixes #16148 

#### Change overview
- Updated ZAP files that create the structure classes to include a description function. 
- Update darwin-framework-tool to iterated and log 

#### Testing
- Validated that the darwin-framework-tool display values in classes, even within a list.
closed 2022-06-02 by woody-apple