project-chip/connectedhomeip #16154: Chip Tool: No support for reading the "Standard Namespace" attribute in the "Mode Select" cluster

dwgaul, 2022-03-12:
Noticed that the Chip tool does not support reading the "Standard Namespace" attribute in the "Mode Select" cluster:

```
 ./chip-tool modeselect read standard-namespace 0x000000000001B675 1 
[1647045385813] [61037:1286032] CHIP: [TOO] Unknown attribute: standard-namespace
Usage:
  ./chip-tool modeselect read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * current-mode                                                                      |
  | * supported-modes                                                                   |
  | * on-mode                                                                           |
  | * start-up-mode                                                                     |
  | * description                                                                       |
  | * server-generated-command-list                                                     |
  | * client-generated-command-list                                                     |
  | * attribute-list                                                                    |
  | * feature-map                                                                       |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
```

Not sure if this was intentional, but it does block attribute testing in certification.

somu1710, 2022-03-18:
It's caught chaitanya,I have already raised a ticket also.I have the ticket for standard namespace attribute.

bzbarsky-apple, 2022-03-28:
Attribute wasn't in the cluster XML.  Added in https://github.com/project-chip/connectedhomeip/pull/15888

closed 2022-03-28 by bzbarsky-apple