project-chip/connectedhomeip #16155: 2nd chiptool app reporting settings overrides 1st chiptool reporting settings

kean-apple, 2022-03-12:
2nd chiptool app reporting settings overrides 1st chiptool reporting settings

Use case: user adds Matter device to one Ecosystem (which enables reporting at interval X). Later, adds a 2nd Ecosystem (which enables reporting Y)..I would expect to still get reporting for both ecosystems apps and the different rates.

Initial: load 2 iOS chiptool apps on iPhone

1. From CHIPtool 1, pair M5 board
2. Enable commissioning window (enable pairing settings on CHIPtool 1) to add new fabric for CHIPTool 2 
3. From CHIPtool 1, set reporting settings to min:1s, max:2s, delta 1C - verify updates to M5 board temp sensor reflects on Chiptool 1 after 1s (Note: need to wait 30s for reporting to get enabled before changing temp sensor)
4. From CHIPtool 2, set reporting settings to min:30s, max:60s, delta 5C -verify updates to M5 board temp sensor reflects on Chiptool 2 after 30s (Note: need to wait 30s for reporting to get enabled before changing temp sensor)

In Step 4, M5 board is reporting at 30s interval. CHIPtool1 does no reflect temp changes from M5 board anymore

Expected: CHIPTool 1 should still get reporting updates

kean-apple, 2022-03-12:
[2nd-chip-tool-reporting-stops-1st-reporting-M5logs.txt](https://github.com/project-chip/connectedhomeip/files/8236691/2nd-chip-tool-reporting-stops-1st-reporting-M5logs.txt)

kean-apple, 2022-03-12:
[2nd-chip-reporting-stops-1st-reporting-CHIPtool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8236702/2nd-chip-reporting-stops-1st-reporting-CHIPtool-logs.txt)

bzbarsky-apple, 2022-03-12:
So I tried this locally...  A subscription with minInterval 1s and maxInterval 1s to my m5stack keeps dying after going for a bit.  How long it takes to do this is somewhat random, but I expect any time any packet gets lost for any reason it happens, since the MRP idle retry interval is 5000ms and the active retry interval is 2000ms on Darwin...

kean-apple, 2022-03-14:
Actually, trying this again on SHA d86a7d49e050f6d9eec438934ea205aa93ce77fd

It seems like when you switch away from CHIPtool 1 Temp sensor UI page, the reporting from the M5 board stops.. even without setting the new reporting interval on the chiptool2. I see this message on M5 board
[switching-away-from-chiptool1-reporting-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8248167/switching-away-from-chiptool1-reporting-M5-logs.txt)

E (1334867) chip[EM]: Flushed pending ack for MessageCounter:10101996 on exchange 56720i
E (1334877) chip[DL]: Long dispatch time: 126 ms, for event type 3
E (1335677) chip[DMG]: Unblock report hold after min 1 seconds
E (1336677) chip[DMG]: Refresh subscribe timer sync after 1 seconds
E (1336677) chip[DMG]: <RE> Sending report (payload has 15 bytes)...
E (1336677) chip[DMG]: IM RH moving to [AwaitingReportResponse]
E (1336697) chip[IN]: Prepared secure message 0x3ffca184 to 0x4DF3646480E54C20 (1)  of type 0x5 and protocolId (0, 1) on exchange 56721i with MessageCounter:4674529.
E (1336707) chip[IN]: Sending encrypted msg 0x3ffca184 with MessageCounter:4674529 to 0x4DF3646480E54C20 (1) at monotonic time: 000000000014623F msec
E (1336717) chip[DMG]: Refresh Subscribe Sync Timer with max 2 seconds
E (1336727) chip[DMG]: <RE> ReportsInFlight = 1 with readHandler 0, RE has no more messages
E (1337727) chip[DMG]: Unblock report hold after min 1 seconds
E (1338727) chip[DMG]: Refresh subscribe timer sync after 1 seconds
I (1339637) ROUTE_HOOK: Received RIO
I (1339637) ROUTE_HOOK: prefix FDB7:92FF:3249:: lifetime 1800

E (1341717) chip[EM]: Retransmitting MessageCounter:4674529 on exchange 56721i Send Cnt 0
E (1341727) chip[IN]: Sending encrypted msg 0x3ffca184 with MessageCounter:4674529 to 0x4DF3646480E54C20 (1) at monotonic time: 00000000001475D8 msec
E (1343727) chip[EM]: Retransmitting MessageCounter:4674529 on exchange 56721i Send Cnt 1
E (1343727) chip[IN]: Sending encrypted msg 0x3ffca184 with MessageCounter:4674529 to 0x4DF3646480E54C20 (1) at monotonic time: 0000000000147DAA msec
E (1345727) chip[EM]: Retransmitting MessageCounter:4674529 on exchange 56721i Send Cnt 2
E (1345727) chip[IN]: Sending encrypted msg 0x3ffca184 with MessageCounter:4674529 to 0x4DF3646480E54C20 (1) at monotonic time: 000000000014857C msec
E (1347727) chip[EM]: Failed to Send CHIP MessageCounter:4674529 on exchange 56721i sendCount: 3 max retries: 3
E (1348697) chip[DMG]: Time out! failed to receive status response from Exchange: 56721i
E (1348697) chip[DMG]: <RE> OnReportConfirm: NumReports = 0
[switching-away-from-chiptool1-reporting-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8248164/switching-away-from-chiptool1-reporting-chiptool-logs.txt)

bzbarsky-apple, 2022-03-16:
> It seems like when you switch away from CHIPtool 1 Temp sensor UI page, the reporting from the M5 board stops

@kean-apple Is the switch to a different app on the same phone?  Or just a different part of the same app?

kean-apple, 2022-03-16:
@bzbarsky-apple changing to a different screen in CHIPtool 1 eg. back to top level menu screen causes the reporting to stop

bzbarsky-apple, 2022-03-16:
@kean-apple But actually stop (as in drop subscription), not the symptoms observed in https://github.com/project-chip/connectedhomeip/issues/16198 where the reports are still coming in but just not updating the CHIPTool UI?

kean-apple, 2022-03-16:
right, in this case the reporting just stops

kean-apple, 2022-03-17:
With SHA 4f22a7935b62bde8b36d921cff75714aeca001e1 this is kinda working..albeit a bit slow

 I'm not seeing reporting stop when going to top level UI or backgrounding CHIPtool 1 0 fixed by https://github.com/project-chip/connectedhomeip/pull/16253

Here is what I'm seeing:
1. Create reporting interval on CHIPtool 1 - 1s/2s/1c delta - reporting working
2. Background Chiptool1 and switch to CHIP2 - still seeing reporting in background
3. On Chiptool 2, set reporting interval to 30s/60s/5c delta - I see the 1 sec reporting stop, 30s reporting kicks in
4. Switch back to CHIPtool 1 - I'm subscribing for reporting happening, wait 30s and I'm seeing the 1s reporting kick in again.
5. Switch back to CHIPtool 2 - See 1 sec reporting stop, have to wait a while eg. 60s and I'm subscribing for reporting happening, wait 30s and I'm seeing the 30s reporting kick in again

kean-apple, 2022-03-17:
[reporting-2chiptools-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8287204/reporting-2chiptools-M5-logs.txt)
[reporting-2chiptools-chiptool-logs.txt.zip](https://github.com/project-chip/connectedhomeip/files/8287205/reporting-2chiptools-chiptool-logs.txt.zip)

kean-apple, 2022-03-17:
For steps 4 & 5, just waiting a while eg. at least 30s without needing to tap on send reporting button

kean-apple, 2022-03-19:
Able to verify 2 different reporting intervals working when using 2 chip tools running a separate devices

closed 2022-05-22 by bzbarsky-apple