project-chip/connectedhomeip #16163: [TC_NC_1.1] Feature map is implememted but while passing this command we are getting 0x86(Unsupported attribute)

sowmyassp, 2022-03-14:
#### Problem
Feature map is implememted but while passing this command we are getting 0x86(Unsupported attribute)
 * expected behavior :  FeatureMap value should be 1
 * actual behavior : Response Failure: IM Error 0x00000586: General error: 0x86
 * steps to reproduce : ./chip-tool descriptor read feature-map 1 1
 * screenshots, images, logs, etc.
 >
[TC_NC_1.1_feature.txt](https://github.com/project-chip/connectedhomeip/files/8242836/TC_NC_1.1_feature.txt)

sowmyassp, 2022-03-18:
Same issue facing in [TC-BOOL-1.1] test-case also

sowmyassp, 2022-03-22:
[TC-NC-1.3]Feature map is implememted but while passing this command we are getting 0x86(Unsupported attribute)

expected behavior : FeatureMap value should be 2
actual behavior : Response Failure: IM Error 0x00000586: General error: 0x86
steps to reproduce : ./chip-tool descriptor read feature-map 1 1
screenshots, images, logs, etc.
[TC_NC_1.3.txt](https://github.com/project-chip/connectedhomeip/files/8321971/TC_NC_1.3.txt)

bzbarsky-apple, 2022-03-28:
> Feature map is implememted 

Is it?  In the SDK descriptor cluster does not have a feature map attribute.

sowmyassp, 2022-03-29:
@bzbarsky-apple Sorry actually the command should be "./chip-tool networkcommissioning read feature-map 1 0" but I read feature-map for descriptor, sorry its my mistake

bzbarsky-apple, 2022-03-30:
@sowmyassp OK.  Does reading the networkcommissioning feature map work for you?  I would expect that to work.

sowmyassp, 2022-03-30:
@bzbarsky-apple Yes its working

closed 2022-04-04 by sowmyassp