project-chip/connectedhomeip #16167: [TE8][TC_WNCV_4_1] Chip-Tool truncates 100th data

eve-cxrp, 2022-03-14:
#### Problem
Step 4a from test case verification steps:

`./chip-tool windowcovering go-to-lift-percentage 75 7520 1 1`

chip-tool should send `7520` as target lift percentage 100th but it truncates it to 7500.

jmeg-sfy, 2022-03-14:
this is a bug on the cluster window covering SDK
need to a do PR for this

eve-cxrp, 2022-03-15:
I don't think it is? I was verifying it by calling directly `WindowCovering::Attributes::TargetPositionLiftPercent100ths::Get` and this function was already returning me 7500. This had nothing to do with the `window-covering-server` impl. So my guess was that it is related to `chip-tool` that it uses the `75` as also the value for the `Percent100ths`?

PR #16841 (2022-03-31): WindowCovering: fixes + simplification
#### Problem
What is being fixed?  Examples:
* Fix #16817
* Fix #16626
* Fix #16396
* Fix #16167
* Fix #16007
* All those are related to Current -> not equals target
* or GoTos not using the right parameter

#### Change overview
We do 3 fixes
- EFR32 remove overwriting of current positions val at preset
- EFR32 fix Current == Target at the end of motion
- ALL: GoTos are fixed with the right input param

Otherwise there are some factorization to simplify the code a bit
for the EFR32 Demo

#### Testing
How was this tested? (at least one bullet point require [... cut]
closed 2022-03-31 by bzbarsky-apple