project-chip/connectedhomeip #16178: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort

caipiblack, 2022-03-14:
#### Problem

Many times I got a Bluez failure during the commissiong of BLE-THREAD devices. 

```
GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
```

My Matter controller is running on OpenWRT but we also have the problem on Raspbian.

```
[1647271250.955612][21062:21080] CHIP:SC: Assigned local session key ID 7
[1647271250.955713][21062:21080] CHIP:SC: Including MRP parameters in PBKDF param request
[1647271250.955778][21062:21080] CHIP:IN: Prepared plaintext message 0xf64004bc to 0x0000000000000000 (0)  of type 0x20 and protocolId (0, 0) on exchange 13999i with MessageCounter:1080752132.
[1647271250.955816][21062:21080] CHIP:IN: Sending plaintext msg 0xf64004bc with MessageCounter:1080752132 to 0x0000000000000000 at monotonic time: 6899927 msec
[1647271250.955835][21062:21080] CHIP:IN: Message appended to BLE send queue
[1647271250.955846][21062:21080] CHIP:SC: Sent PBKDF param request
TX: {"id":1,"jsonrpc":"2.0","result":{"status":0}}
[1647271250.963530][21062:21071] CHIP:BLE: BLE removing known devices.
[1647271250.964839][21062:21071] CHIP:BLE: BLE initiating scan.
[1647271251.245211][21062:21071] CHIP:BLE: Device F6:7B:B8:6D:B3:6D does not look like a CHIP device.
[1647271251.707264][21062:21071] CHIP:BLE: Device CC:37:9E:4F:13:BD does not look like a CHIP device.
[1647271252.097699][21062:21071] CHIP:BLE: Device F6:7B:B8:6D:B3:6D does not look like a CHIP device.
[1647271252.561688][21062:21071] CHIP:BLE: Device CC:37:9E:4F:13:BD does not look like a CHIP device.
[1647271252.843267][21062:21071] CHIP:BLE: Device C6:60:14:80:57:35 does not look like a CHIP device.
[1647271252.959687][21062:21071] CHIP:BLE: Device F6:7B:B8:6D:B3:6D does not look like a CHIP device.
[1647271253.269497][21062:21071] CHIP:BLE: New device scanned: 00:60:37:34:CC:38
[1647271253.269538][21062:21071] CHIP:BLE: Device discriminator match. Attempting to connect.
[1647271253.272818][21062:21071] CHIP:BLE: Scan complete notification without an active scan.
[1647271254.030765][21062:21071] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
[1647271254.031361][21062:21062] CHIP:DL: HandlePlatformSpecificBLEEvent 16386
[1647271254.031385][21062:21062] CHIP:IN: Clearing BLE pending packets.
[1647271254.031409][21062:21062] CHIP:IN: BleConnection Error: ../../src/platform/Linux/bluez/Helper.cpp:1738: CHIP Error 0x000000AC: Internal error
[1647271280.983406][21062:21062] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
TX: {"jsonrpc":"2.0","method":"onPairingStatusUpdate","params":{"status":2}}
[1647271280.983700][21062:21062] CHIP:TOO: Secure Pairing Failed
```

This problem has been reported in many issues but they has been closed without real fix:

https://github.com/project-chip/connectedhomeip/issues/9623
https://github.com/project-chip/connectedhomeip/issues/8407
https://github.com/project-chip/connectedhomeip/issues/8416

Restarting the Bluetooth service or playing with RFKill to turn off/on the BLE dongle works but (i think) is not a solution. 

Did someone know the root cause of this problem ? How to fix it ?

RudyGOGOGO, 2022-04-04:
@caipiblack I am working on how to cross compile the Matter Controller for OpenWrt however I get no luck.
I luckily find this issue you raised. Would you mind providing some hint or the GN command about how you cross compiled the Controller for OpenWrt? I have struggled with this problem for a whole week, I would really appreciate your insight and your precious time!!!!

Here is the basic info from my side:
1)I get the cross-compiling environment of OpenWrt ready
2)I can cross compile some simple code running in the router which is flashed with OpenWrt
3)I did not find helpful doc about how to set the parameters for GN Cross Compiling, however only GN build scripts are provided in the CHIP-tool doc: https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool

caipiblack, 2022-04-06:
@RudyGOGOGO : 

Two options:

1. Use a CMake project, and in this CMake use `ExternalProject_Add()` to run GN and then to run Ninja. (I'm not using this method because we also have to execute "scripts/activate.sh" before. 
2. Build the Matter libraries and tools externaly using the Yocto/OpenWRT toolchain, here is an example:

```
cd ~/connectedhomeip
rm -rf out/aarch32
. scripts/activate.sh

export SDKTARGETSYSROOT=/opt/arm-scos-linux-gnueabi/sysroot
PKG_CONFIG_LIBDIR=$SDKTARGETSYSROOT/lib/pkgconfig:$SDKTARGETSYSROOT/usr/lib/pkgconfig:$SDKTARGETSYSROOT/usr/share/pkgconfig \
gn gen out/aarch32 --args='target_os="linux" target_cpu="arm" arm_arch="armv7-a"
        import("//build_overrides/build.gni")
        target_cflags=[ "--sysroot='${SDKTARGETSYSROOT}'", "'-Wno-conversion'", "'-Wno-error'", "'-I/opt/arm-scos-linux-gnueabi/sysroot/usr/include/glib-2.0'", "'-I/opt/arm-scos-linux-gnueabi/sysroot/usr/lib/glib-2.0/include'", "-DG_SOURCE_FUNC(f)=((GSourceFunc)(void (*)(void)) (f))" ]
        target_ldflags = [ "--sysroot='${SDKTARGETSYSROOT}'", "'-lz'", "'-lffi'", "'-lgmodule-2.0'" ]
        custom_toolchain="${build_root}/toolchain/custom"
        target_cc="/opt/bin/arm-scos-linux-gnueabi-gcc"
        target_cxx="/opt/bin/arm-scos-linux-gnueabi-g++"
        target_ar="/opt/bin/arm-scos-linux-gnueabi-ar"'
ninja -C out/aarch32
```

_Forget about the G_SOURCE_FUNC macro, this is because I am not using the correct version of GLIB-2.0_

Also, note that currently the GN configuration does not provide install step [...] And I think it's not the priority of the people working on Matter (See here: https://github.com/project-chip/connectedhomeip/issues/1513), so you also have to make a script to export the libraries, headers and binaries that you need to use (or just copy it by hand...)

You also have documentation about cross-compilation here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/nxp_imx8m_linux_examples.md

If you have more questions, please create an issue and ask it in your issue. (To keep this issue focused on the Bluez problem)

YuriChen1028, 2022-05-08:
Hi,
Have you solve this error:
> GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
I have this error as you when i use PRI 4 to commision my device.

YuriChen1028, 2022-05-17:
You can shutdown wifi on your rpi 4b and use Ethernet. Then the commissioning by ble will be ok.

caipiblack, 2022-06-10:
I understand, this problem has been reported in many and many issues, all the answers converge towards the same direction: it is necessary to deactivate the wifi probably to reduce the risk of a packet loss due to WiFi/Bluetooth co-existance. 

If the problem is really due to a packet loss then it means that the protocol/stack is possibly not robust against this.

This error is one of the most frequent during the commissioning. Currently when it happens in our app, the user have to restart the commissioning process and on the next try it works. It's ok but..

resghst, 2022-11-08:
Hi
I faced the same issue on raspberry pi 4, and this problem can easily occur and cause the commissioning to fail. (about 30%-50%)
Even I restart Bluetooth and reboot raspberry pi 4, this error can't solve.
Can anybody help me?

By the way, #9623 also does not work on my raspberry pi 4
sudo btmgmt -i hci0 power off
sudo btmgmt -i hci0 bredr off
sudo btmgmt -i hci0 power on

caipiblack, 2022-11-08:
This problem is a real problem and it’s not only on raspberry pi.

We have the problem on various linux boards:
- On internet gateways running OpenWRT
- On imx8mn boards
- On raspberry pi

Everything that use bluez have the problem.

closed 2023-05-18 by stale[bot]

willyan2002, 2024-06-20:
The Raspberry Pi running the chip tool must have WiFi turned off, otherwise it will cause serious interference to the BLE. You can discover the matter device, but cannot successfully establish the BLE connection. The error log is as you can see.

If your Raspberry Pi‘s Wi-Fi is indeed on, please use nmcli or rfkill to turn it off, and then try to commission some matter device , you may see the success rate is almost 100%.

Raspberry Pi Wi-Fi on
`[1718863192.902905][56656:56657] CHIP:BLE: Device 35:B0:FD:18:C2:C8 does not look like a CHIP device.
[1718863192.916208][56656:56657] CHIP:BLE: New device scanned: C5:33:98:4C:92:A7
[1718863192.916288][56656:56657] CHIP:BLE: Device discriminator match. Attempting to connect.
[1718863192.920537][56656:56657] CHIP:BLE: ChipDeviceScanner has stopped scanning!
[1718863196.772053][56656:56657] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)
[1718863196.772156][56656:56657] CHIP:DL: ConnectDevice retry: 1 out of 4
[1718863197.784491][56656:56657] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)
[1718863197.784578][56656:56657] CHIP:DL: ConnectDevice retry: 2 out of 4
[1718863201.824495][56656:56657] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)
[1718863201.824585][56656:56657] CHIP:DL: ConnectDevice retry: 3 out of 4
[1718863204.335772][56656:56657] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)
[1718863204.335865][56656:56657] CHIP:DL: ConnectDevice retry: 4 out of 4
[1718863204.348656][56656:56657] CHIP:BLE: Device connection failed: src/platform/Linux/bluez/BluezEndpoint.cpp:657: CHIP Error 0x000000AC: Internal error`
Wi-Fi off
ubuntu@ubuntu:$ sudo nmcli radio wifi off
ubuntu@ubuntu:$

`1718863407.072943][57126:57127] CHIP:BLE: Device B0:CE:18:DE:3E:7D does not look like a CHIP device.
[1718863407.093220][57126:57127] CHIP:BLE: Device 51:44:C6:3C:5F:8D does not look like a CHIP device.
[1718863407.106942][57126:57127] CHIP:BLE: Device 62:86:FD:E0:79:93 does not look like a CHIP device.
[1718863407.108374][57126:57127] CHIP:BLE: New device scanned: E3:FC:A3:A8:7A:28
[1718863407.108440][57126:57127] CHIP:BLE: Device discriminator match. Attempting to connect.
[1718863407.113574][57126:57127] CHIP:BLE: ChipDeviceScanner has stopped scanning!
**[1718863407.370060][57126:57127] CHIP:DL: ConnectDevice complete
[1718863407.370156][57126:57127] CHIP:BLE: New device connected: E3:FC:A3:A8:7A:28**
[1718863408.769827][57126:57127] CHIP:DL: Char /org/bluez/hci0/dev_E3_FC_A3_A8_7A_28/service000a on service /org/bluez/hci0/dev_E3_FC_A3_A8_7A_28/service000a`

caojun97, 2024-08-07:
> #### Problem
> Many times I got a Bluez failure during the commissiong of BLE-THREAD devices.
>
> ```
> GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
> ```
>
> My Matter controller is [...]

hello  could you solve the problem?  

When i run the command: `chip-tool pairing ble-wifi 0x7283 MY_SSID MY_PASSWD 20202021 3840`, i got the same problem
```
CHIP:BLE: New device scanned: CA:D8:AE:D3:8D:A2
[1722950997.699406][62794:62795] CHIP:BLE: Device discriminator match. Attempting to connect.
[1722950997.803120][62794:62795] CHIP:BLE: ChipDeviceScanner has stopped scanning!
[1722950998.888196][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722950998.888235][62794:62795] CHIP:DL: ConnectDevice retry: 1 out of 4
[1722951001.407642][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722951001.407680][62794:62795] CHIP:DL: ConnectDevice retry: 2 out of 4
[1722951002.937974][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722951002.938006][62794:62795] CHIP:DL: ConnectDevice retry: 3 out of 4
[1722951003.927847][62794:62795] CHIP:DL: FAIL: ConnectDevice: GDBus.Error:org.bluez.Error.Failed: Software caused connection abort (36)
[1722951003.927895][62794:62795] CHIP:DL: ConnectDevice retry: 4 out of 4
[1722951003.959837][62794:62795] CHIP:BLE: Device connection failed: src/platform/Linux/bluez/BluezEndpoint.cpp:657: CHIP Error 0x000000AC: Internal error
[1722951003.959904][62794:62796] CHIP:DL: HandlePlatformSpecificBLEEvent 16386
[1722951003.959931][62794:62796] CHIP:DIS: Closing all BLE connections
[1722951003.959961][62794:62796] CHIP:TOO: Pairing Failure: src/platform/Linux/bluez/BluezEndpoint.cpp:656: CHIP Error 0x000000AC: Internal error
[1722951003.960305][62794:62794] CHIP:CTL: Shutting down the commissioner
[1722951003.960342][62794:62794] CHIP:CTL: Shutting down the controller
[1722951003.960385][62794:62794] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1722951003.960404][62794:62794] CHIP:FP: Forgetting fabric 0x1
[1722951003.960434][62794:62794] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1722951003.960562][62794:62794] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1722951003.960581][62794:62794] CHIP:TS: Reverted Last Known Good Time to previous value
[1722951003.960622][62794:62794] CHIP:CTL: Shutting down the commissioner
[1722951003.960675][62794:62794] CHIP:CTL: Shutting down the controller
[1722951003.960713][62794:62794] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1722951003.960874][62794:62794] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1722951003.960934][62794:62794] CHIP:FP: Shutting down FabricTable
[1722951003.960954][62794:62794] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1722951003.961037][62794:62794] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1722951003.961051][62794:62794] CHIP:TS: Reverted Last Known Good Time to previous value
[1722951003.961255][62794:62794] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-e07VRm)
[1722951003.961650][62794:62794] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1722951003.961680][62794:62794] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1722951003.961691][62794:62794] CHIP:DL: Inet Layer shutdown
[1722951003.961700][62794:62794] CHIP:DL: BLE shutdown
[1722951003.963304][62794:62794] CHIP:DL: System Layer shutdown
[1722951003.963765][62794:62794] CHIP:TOO: Run command failure: src/platform/Linux/bluez/BluezEndpoint.cpp:656: CHIP Error 0x000000AC: Internal error

```

So why i can get the problem, i didn't know.  Did you sovle it?  Looking forward to your reply

caipiblack, 2024-08-07:
> hello could you solve the problem?
>
> When i run the command: `chip-tool pairing ble-wifi 0x7283 MY_SSID MY_PASSWD 20202021 3840`, i got the same problem
>
> ```
> CHIP:BLE: New device scanned: [...]

Hello,

No, i didn’t solved this error.
Many people say it’s only on raspberry pi due to wifi and ble co existance.

but in fact we have the problem in all the linux devices where we integrate matter (openwrt gateway, yocto hub, linux devices in general)

for me it’s on every device that has bluez

But now we don’t reproduce the problem because we change how the commissioning is made:
- We do the first commissioning using a mobile app now(so the ble)
- And after we commission the device on network in the linux device using opencommissiongwindow
- In other terms now we do the « ble part » in mobile phone and we don’t have the issue because maybe it’s better implemented in mobile phones…

wahadatjan, 2024-10-25:
Using external BLE adapter can solve the problem