project-chip/connectedhomeip #16186: Backoff behavior for auto-resubscribe seems buggy.

bzbarsky-apple, 2022-03-14:
#### Problem
The way auto-resubscribe works is that ReadClient keeps track of number of past subscribe attempts and when doing a resubscribe backs off based on how many there have been.

The problem is that when we succesfully establish a subscription this counter is not cleared.  So if we subscribe, then two days later have a drop and resubscribe, then two days later have another drop, etc, the backoff will keep growing after each drop even though there's no reason to do that.

 #### Proposed Solution
We should only do backoff if the subscribe attempt fails (which makes sense to do).  Once it succeeds, we should clear the counter so next time there's a drop we try to resubscribe quickly.

mrjerryjohns, 2022-03-14:
Yeah, that's a bug. We should fix that to clear the counter back to 0 on successful establishment of a subscription.

mrjerryjohns, 2022-03-14:
Specifically, that looks to me `mNumRetries`.

bzbarsky-apple, 2022-03-15:
Fixed by https://github.com/project-chip/connectedhomeip/pull/16199

closed 2022-03-15 by bzbarsky-apple