project-chip/connectedhomeip #16205: TE8 Mac Chip Tool Darwin: Missing support for all optional attributes in the "Pressure Measurement" cluster

dwgaul, 2022-03-15:
Chip-tool-darwin does not support reading all optional attributes:

<details><summary>Chip-tool-darwin supported attributes</summary>

```
./chip-tool-darwin pressuremeasurement read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * measured-value                                                                    |
  | * min-measured-value                                                                |
  | * max-measured-value                                                                |
  | * attribute-list                                                                    |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
```
</details>

<details><summary>Chip-tool supported attributes</summary>

```
 ./chip-tool pressuremeasurement read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * measured-value                                                                    |
  | * min-measured-value                                                                |
  | * max-measured-value                                                                |
  | * tolerance                                                                         |
  | * scaled-value                                                                      |
  | * min-scaled-value                                                                  |
  | * max-scaled-value                                                                  |
  | * scaled-tolerance                                                                  |
  | * scale                                                                             |
  | * server-generated-command-list                                                     |
  | * client-generated-command-list                                                     |
  | * attribute-list                                                                    |
  | * feature-map                                                                       |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
```
</details>

Compared to the plain chip-tool, chip-tool-darwin is missing support for the following attributes: 
- tolerance 
- scale 
- scaled-value 
- min-scaled-value 
- max-scaled-value 
- scaled-tolerance

Since these are optional, technically chip-tool-darwin passes TC-PRS-3-1 cert test case. There is no requirement forcing controllers/clients to support optional attributes.

Filing this to track if this is intended behavior for chip-tool-darwin.

bzbarsky-apple, 2022-03-28:
@krypton36

closed 2022-10-03 by stale[bot]

bzbarsky-apple, 2022-10-06:
For what it's worth, this is just fixed now.