project-chip/connectedhomeip #16209: Crash trying to call OpenCommissioningWindow for a node ID that can't be connected to

bzbarsky-apple, 2022-03-15:
#### Problem
Crash with this stack:
```
0   libsystem_kernel.dylib        	       0x1850ed270 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x185181474 pthread_kill + 272
2   libsystem_c.dylib             	       0x185053474 abort + 164
3   CHIP                          	       0x1176f1c40 chipAbort + 20
4   CHIP                          	       0x116325fa8 chip::Optional<chip::SessionHandle>::Value() const & + 120
5   CHIP                          	       0x116b2a040 chip::ChipError chip::Controller::ClusterBase::InvokeCommand<chip::app::Clusters::AdministratorCommissioning::Commands::OpenCommissioningWindow::Type>(chip::app::Clusters::AdministratorCommissioning::Commands::OpenCommissioningWindow::Type const&, void*, void (*)(void*, chip::app::Clusters::AdministratorCommissioning::Commands::OpenCommissioningWindow::Type::ResponseType const&), void (*)(void*, chip::ChipError), chip::Optional<unsigned short> const&) + 272
6   CHIP                          	       0x1176e68c4 chip::Controller::DeviceController::OpenCommissioningWindowInternal() + 736
7   CHIP                          	       0x1176e7300 chip::Controller::DeviceController::OpenCommissioningWindowWithCallback(unsigned long long, unsigned short, unsigned int, unsigned short, chip::Controller::DeviceController::CommissioningWindowOption, chip::Callback::Callback<void (*)(void*, unsigned long long, chip::ChipError, chip::SetupPayload)>*, bool) + 652
8   CHIP                          	       0x11661edd0 chip::Controller::DeviceController::OpenCommissioningWindow(unsigned long long, unsigned short, unsigned int, unsigned short, chip::Controller::DeviceController::CommissioningWindowOption, chip::SetupPayload&) + 116
```

 #### Proposed Solution
The basic problem is the use of `FindExistingSession` in `DeviceController::OpenCommissioningWindowWithCallback`, `DeviceController::OpenCommissioningWindowInternal`.  The caller in this case tried to connect before and failed, so now we have an `OperationalDeviceProxy` hanging around in an un-connected state.  Then we find it via `FindExistingSession`, try to use it to send a command, and crash because it does not in fact have a secure session attached to it.

I looked at all other consumers of `FindExistingSession` and none of the rest of them expect a non-null CASE session.  It's just this one that's broken.

Some ways I see of fixing this:
1) Use `GetConnectedDevice` here, but that leaves the problem of the controller only being able to do this on one node at a time, without ways to report errors, etc.
2) Change the API to pass in a connected OperationalDeviceProxy (i.e. make consumer call `GetConnectedDevice`).  Same issues here, but might be simpler to implement, maybe.
3) Move this functionality to OperationalDeviceProxy.  The consumer would be required to `GetConnectedDevice` itself, and all the state would live in `OperationalDeviceProxy`.
4) Move this functionality to a helper class thing that gets a connected `OperationalDeviceProxy` and uses it to do what these APIs are currently doing.
5) Move this functionality to some helper class thing that gets a node id and controller reference and uses that to `GetConnectedDevice`, etc, etc.

If we want to do option 4, we can probably start with option 2 and convert all consumers to `GetConnectedDevice` and then do the splitting out into a separate class as a followup.

If we want to do option 5, we probably need fewer changes in the consumers.

bzbarsky-apple, 2022-03-15:
@austinh0 Which of the above would work best for the Android consumers of these APIs?

@yufengwangca Similar question for the callsite in `ChipDeviceController-ScriptBinding.cpp`.

austinh0, 2022-03-15:
In Android code, almost all functions already take in a `DeviceProxy` pointer instead of a node id, so options that use `GetConnectedDevice` separately (2, 3) would be pretty simple to use for us.

yufengwangca, 2022-03-16:
I prefer option 4, but GetConnectedDevice does not return a connected device, just try to establishes a secure connection with it

closed 2022-03-24 by bzbarsky-apple