project-chip/connectedhomeip #16213: [TE8] Instable chip-tool (bluez madness)

eve-cxrp, 2022-03-15:
#### Problem
chip-tool is very unstable during the `ble-thread` pairing and it is shere torture to execute test cases. For me this happens 70% of the time with one of the following errors:

[bluez1.log](https://github.com/project-chip/connectedhomeip/files/8251257/bluez1.log)
[bluez2.log](https://github.com/project-chip/connectedhomeip/files/8251258/bluez2.log)
[bluez3.log](https://github.com/project-chip/connectedhomeip/files/8251259/bluez3.log)

also related issue: https://github.com/project-chip/connectedhomeip/issues/16178

eve-cxrp, 2022-03-15:
Some suggested in the slack channel that `sudo rfkill block wifi` or `sudo ifconfig wlan0 down` which had 0 effect on my setup. I also disabled the wifi module at boot-up with:

```
ubuntu@matter-dm:~/new$ cat /boot/firmware/usercfg.txt
dtoverlay=disable-wifi
```

eve-cxrp, 2022-04-06:
related issue with more infos: https://github.com/CHIP-Specifications/chip-certification-tool/issues/324

caipiblack, 2022-04-06:
Any news about this problem ?

_It is known issue since TE5 (and possibly more)_

@eve-cxrp : Which Bluetooth dongle was you using ? 

I am using nRF52840 with USB<>HCI firmware.

I don't know if the problem is Bluez, Hardware or Matter.

- For people using chip-tool it's probably "ok" because after every request chip-tool restart and sometimes it works.
- For people using the stack, on TE8 tag if a pairing fail, then the next pairing will fail (because of another issue), so this issue is a real pain !

closed 2022-10-18 by stale[bot]