project-chip/connectedhomeip #16220: [SQA] TC-I-2.3 Doesnt reproduce expected outcome in thread connectivity.

Survensa, 2022-03-15:
**Expected outcome** : DUT executes a blink effect.
**Actual outcome** : No blink effect is reflected. 

**Commit used** : 2982c0ca0623773327def4cc7180d07f6b959c42

**Command used** : `./chip-tool identify trigger-effect 0 0 92 1`

**Output DUT log** :
 
> I: 19990146 [EM]Received message of type 0x30 with protocolId (0, 0) and MessageCounter:1962765323 r
D: 19990157 [EM]Handling via exchange: 61266r, Delegate: 0x20003d54                                         
I: 19990163 [IN]CASE Server received Sigma1 message. Starting handshake. EC 0x20005210                      
I: 19990170 [DIS]CASE handshake initiated, closing all BLE Connections                                      
D: 19990177 [SC]Waiting for Sigma1 msg                                                                      
I: 19990180 [IN]CASE Server disabling CASE session setups                                                   
D: 19990185 [SC]Received Sigma1 msg                                                                         
D: 19990189 [SC]Found MRP parameters in the message                                                         
D: 19990193 [SC]Peer assigned session key ID 1                                                              
I: 19990197 [DIS]Fabric (1) loaded from storage. Calling OnFabricRetrievedFromStorage                       
I: 19990205 [DIS]Fabric (1) loaded from storage. Calling OnFabricRetrievedFromStorage                       
I: 19990213 [ZCL]OpCreds: Fabric 0x1 was retrieved from storage. FabricId 0x0000000000000001, NodeId 0x00001
I: 19990225 [ZCL]OpCreds: Call to fabricListChanged                                                         
D: 19990229 [DMG]Endpoint 0, Cluster 0x0000_003E update version to d2bfd45d                                 
D: 19990236 [DMG]Endpoint 0, Cluster 0x0000_003E update version to d2bfd45e                                 
D: 19990243 [IN]Generating DestinationID. Fabric ID 0x0000000000000001, Dest node ID 0x000000000000005C, Raa
D: 19990254 [IN]0x55, 0x5f, 0xd0, 0x0f, 0xee, 0xc3, 0x06, 0x83,                                             
D: 19990260 [IN]0xa3, 0xbd, 0x31, 0x12, 0x0d, 0x1c, 0x32, 0x60,                                             
D: 19990265 [IN]0x2a, 0x6a, 0x7b, 0x83, 0x45, 0x1e, 0x8d, 0x70,                                             
D: 19990271 [IN]0x98, 0x3f, 0x3d, 0xd3, 0x90, 0xc8, 0x57, 0x86,                                             
D: 19990277 [IN]Root pubkey                                                                                 
D: 19990280 [IN]0x04, 0x6d, 0x8c, 0x5c, 0x75, 0x75, 0xd0, 0xce,                                             
D: 19990286 [IN]0x4b, 0x38, 0x3c, 0x3a, 0x5b, 0xcb, 0xa0, 0xc8,                                             
D: 19990292 [IN]0x63, 0xe8, 0x1c, 0x0a, 0xf1, 0x8f, 0xcb, 0x83,                                             
D: 19990297 [IN]0x53, 0x57, 0x69, 0x71, 0x42, 0x49, 0xbd, 0xbf,                                             
D: 19990303 [IN]0x2c, 0x25, 0x47, 0x54, 0xd9, 0x48, 0x65, 0x80,                                             
D: 19990309 [IN]0xec, 0xb8, 0x4c, 0x31, 0xf0, 0x3c, 0x36, 0x07,                                             
D: 19990315 [IN]0xdc, 0x15, 0x38, 0x87, 0x4a, 0xce, 0x5b, 0xd3,                                             
D: 19990321 [IN]0xba, 0x32, 0x4c, 0xae, 0x94, 0xdf, 0x98, 0xa3,                                             
D: 19990327 [IN]0xfc,                                                                                       
D: 19990329 [IN]IPK                                                                                         
D: 19990331 [IN]0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,                                             
D: 19990337 [IN]0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,                                             
D: 19990343 [IN]Generated DestinationID output                                                              
D: 19990347 [IN]0x52, 0x1b, 0x1b, 0x0a, 0x5b, 0x52, 0xfe, 0x2a,                                             
D: 19990353 [IN]0x3c, 0xee, 0x82, 0x01, 0x37, 0xb5, 0x90, 0x52,                                             
D: 19990359 [IN]0x41, 0x75, 0xb7, 0x35, 0xd9, 0x24, 0x54, 0x39,                                             
D: 19990365 [IN]0x52, 0x7f, 0x44, 0xee, 0x65, 0x81, 0x7d, 0x29,                                             
I: 19990370 [DIS]Fabric (1) loaded from storage. Calling OnFabricRetrievedFromStorage                       
I: 19990378 [DIS]Fabric (1) loaded from storage. Calling OnFabricRetrievedFromStorage                       
I: 19990386 [ZCL]OpCreds: Fabric 0x1 was retrieved from storage. FabricId 0x0000000000000001, NodeId 0x00001
I: 19990398 [ZCL]OpCreds: Call to fabricListChanged                                                         
D: 19990402 [DMG]Endpoint 0, Cluster 0x0000_003E update version to d2bfd45f                                 
D: 19990409 [DMG]Endpoint 0, Cluster 0x0000_003E update version to d2bfd460                                 
D: 19990638 [SC]Including MRP parameters                                                                    
I: 19990642 [IN]Prepared unauthenticated message 0x2000512c to 0x0000000000000000 (0)  of type 0x31 and pro.
I: 19990658 [IN]Sending unauthenticated msg 0x2000512c with MessageCounter:3322322244 to 0x0000000000000000c
D: 19990672 [SC]Sent Sigma2 msg                                                                             
I: 19990798 [EM]Received message of type 0x32 with protocolId (0, 0) and MessageCounter:1962765324 on exchar
D: 19990808 [EM]Found matching exchange: 61266r, Delegate: 0x20003d60                                       
D: 19990815 [EM]Rxd Ack; Removing MessageCounter:3322322244 from Retrans Table on exchange 61266r           
D: 19990823 [EM]Removed CHIP MessageCounter:3322322244 from RetransTable on exchange 61266r                 
D: 19990831 [SC]Received Sigma3 msg                                                                         
E: 19990839 [SC]The device does not support GetClock_RealTimeMS() API. This will eventually result in CASE .
D: 19990964 [IN]Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root puy
D: 19990974 [IN]0x04, 0x6d, 0x8c, 0x5c, 0x75, 0x75, 0xd0, 0xce,                                             
D: 19990980 [IN]0x4b, 0x38, 0x3c, 0x3a, 0x5b, 0xcb, 0xa0, 0xc8,                                             
D: 19990985 [IN]0x63, 0xe8, 0x1c, 0x0a, 0xf1, 0x8f, 0xcb, 0x83,                                             
D: 19990991 [IN]0x53, 0x57, 0x69, 0x71, 0x42, 0x49, 0xbd, 0xbf,                                             
D: 19990997 [IN]0x2c, 0x25, 0x47, 0x54, 0xd9, 0x48, 0x65, 0x80,                                             
D: 19991003 [IN]0xec, 0xb8, 0x4c, 0x31, 0xf0, 0x3c, 0x36, 0x07,                                             
[... 7 lines cut ...]
I: 19991112 [IN]Sending unauthenticated msg 0x2000512c with MessageCounter:3322322245 to 0x0000000000000000c
I: 19991126 [IN]CASE Session established. Setting up the secure channel.                                    
D: 19991133 [IN]New secure session created for device 0x000000000001B669, LSID:71 PSID:1!                   
I: 19991142 [IN]CASE secure channel is available now.                                                       
I: 19991147 [IN]CASE Server enabling CASE session setups                                                    
I: 19991156 [EM]Received message of type 0x8 with protocolId (0, 1) and MessageCounter:946999 on exchange 6r
D: 19991167 [EM]Handling via exchange: 61267r, Delegate: 0x200074f8                                         
I: 19991174 [DMG]AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0003 e=1 p=o              
D: 19991182 [DMG]Received command for Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0040                    
I: 19991190 [ZCL]RX identify:trigger effect 0x0 variant 0x0                                                 
D: 19991195 [DMG]ICR moving to [AddingComm]                                                                 
D: 19991199 [DMG]ICR moving to [AddedComma]                                                                 
D: 19991203 [DMG]Decreasing reference count for CommandHandler, remaining 0                                 
I: 19991211 [IN]Prepared secure message 0x20005144 to 0x000000000001B669 (1)  of type 0x9 and protocolId (0.
[... cut ...]
D: 19991274 [EM]Removed CHIP MessageCounter:3322322245 from RetransTable on exchange 61266r                 
I: 19991283 [EM]Received message of type 0x10 with protocolId (0, 0) and MessageCounter:947000 on exchange r
D: 19991293 [EM]Found matching exchange: 61267r, Delegate: (nil)                                            
D: 19991299 [EM]Rxd Ack; Removing MessageCounter:8537887 from Retrans Table on exchange 61267r              
D: 19991307 [EM]Removed CHIP MessageCounter:8537887 from RetransTable on exchange 61267r

andy31415, 2022-04-13:
@Survensa - please add information for developers on reproduction:

- what application to use (light, lock, all-clusters, pump etc) and what platform (nrf, efr, other)

Survensa, 2022-04-19:
@andy31415 
Application : Lightining App
Platform : Thread (nrf)

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @cjandhyala to verify this is still an issue.

cjandhyala, 2022-05-25:
@Survensa confirm if this is still an issue

closed 2022-05-25 by cjandhyala