project-chip/connectedhomeip #16241: Relaunching chip tool while reporting is happening from M5 stops the reporting

vinay-apple, 2022-03-15:
#### Problem
Relaunching chip tool while reporting is happening from M5 stops the reporting 
 
SHA: 494cfb24326a148b27622e89c2aad37cac4e97ce

Steps to reproduce

1. Pair M5 with chip tool and verify light toggle works.
2. Setup reporting for temperature sensor with Min 1s, Max 2s, delta 1 deg c
3. Verify reporting is working
4. relaunch chip tool app and check that reporting stops working. 

Seeing below logs on M5:

E (250801) chip[EM]: Failed to Send CHIP MessageCounter:11932893 on exchange 379i sendCount: 3 max retries: 3
I (251771) chip[DMG]: Time out! failed to receive status response from Exchange: 379i
I (266931) all-clusters-app: Display going to sleep...
> aI (473741) chip[EM]: Received message of type 0x30 with protocolId (0, 0) and MessageCounter:3805453696 on exchange 54028r

[M5-logs-relaunching-chiptool-stops-reporting.txt](https://github.com/project-chip/connectedhomeip/files/8256195/M5-logs-relaunching-chiptool-stops-reporting.txt)

[chiptool-logs-relaunching-chiptool-stops-reporting.txt](https://github.com/project-chip/connectedhomeip/files/8256193/chiptool-logs-relaunching-chiptool-stops-reporting.txt)

bzbarsky-apple, 2022-03-15:
Relaunching chip-tool is expected to drop the subscription.  Subscriptions are not persistent (esp. on clients).

Does it work if you then go back into the temp sensor view and re-establish the subscription, @vinay-apple?

@sagar-apple

vinay-apple, 2022-03-15:
@bzbarsky-apple I need to do "send reporting settings" again for this to starting working again. 

Same behavior is seen even when we background the chip tool. The reporting stops to work, and when we foreground chip tool don't see the reporting working, we need to "send reporting settings" again and then it starts to work. 

Seeing below logs on M5 when we background chip tool while reporting is setup on M5:

E (5161831) chip[EM]: Failed to Send CHIP MessageCounter:7027266 on exchange 2362i sendCount: 3 max retries: 3
I (5162801) chip[DMG]: Time out! failed to receive status response from Exchange: 2362i
I (5179861) chip[EM]: Received message of type 0x1 with protocolId (0, 1) and MessageCounter:7098453 on exchange 2362i
I (5179861) chip[IN]: Prepared secure message 0x3ffe77f4 to 0xD94EF84342B69610 (1)  of type 0x10 and protocolId (0, 0) on exchange 2362i with MessageCounter:7027267.
I (5179881) chip[IN]: Sending encrypted msg 0x3ffe77f4 with MessageCounter:7027267 to 0xD94EF84342B69610 (1) at monotonic time: 00000000004F069D msec
E (5179901) chip[EM]: Dropping unexpected message of type 0x1 with protocolId (0, 1) and MessageCounter:7098453 on exchange 2362i
I (5179901) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:7098454 on exchange 2362i
E (5179921) chip[EM]: OnMessageReceived failed, err = Error CHIP:0x00000070
I (5179931) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:7098455 on exchange 2362i
E (5179941) chip[EM]: OnMessageReceived failed, err = Error CHIP:0x00000070
I (5179941) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:7098456 on exchange 2362i
E (5179961) chip[EM]: OnMessageReceived failed, err = Error CHIP:0x00000070

[M5-logs-background-chiptool-stops-reporting.txt](https://github.com/project-chip/connectedhomeip/files/8256626/M5-logs-background-chiptool-stops-reporting.txt)

[chiptool-logs-background-chiptool-stops-reporting.txt](https://github.com/project-chip/connectedhomeip/files/8256623/chiptool-logs-background-chiptool-stops-reporting.txt)

bzbarsky-apple, 2022-03-16:
Right, that's all expected behavior.  Certainly for relaunching, and looks like for backgrounding too (the backgrounded CHIPTool can't receive packets from the network, so can't get reports and respond to them and then the server drops the subscription).

mrjerryjohns, 2022-03-17:
Closing as "works as expected".

closed 2022-03-17 by mrjerryjohns