project-chip/connectedhomeip #16266: [TE8] TC-TSTAT-3.1 Attributes with client as DUT

cabin15, 2022-03-16:
Test-Case ID:-[TC-TSTAT-3.1] Attributes with client as DUT

Problem
While Executing the behavior of the non-global attributes of the Thermostat cluster, a few attributes throws error code as 0x86(unsupported attribute).
- outdoor-temperature

- occupancy

- pi-cooling-demand

- and so on (Please check with 0x86 error code on attached log)

Test log

andy31415, 2022-05-30:
@cabin15 could you provide a list of attributes based on required/not-required?

For v1.0 we generally mandate only required attributes and are willing to skip over optional ones.

Specifically I see:
  - PICoolingDemand has conformance `[COOL]`
  - Occupancy has conformance `OCC`
  - Outdoor temperature has conformance `O` (so fully optional)
  
  The only attribute that a thermostat **must** support is LocalTemperature.
  
  Beyond that, it should be based on the feature map and optionality.

woody-apple, 2022-06-21:
Spec Issue Review: Given this an older issue, and there is no cert blocker tracking this. Closing. Please re-open if this is still an issue.