project-chip/connectedhomeip #16267: [TE8] [TC-TSAT-3.2] Functionality with client as DUT

cabin15, 2022-03-16:
Test-Case ID:-[TC-TSAT-3.2] Functionality with client as DUT

Problem
While Executing the behavior of the functionality of the Thermostat cluster, a few commands throws error code as 0xc3(unsupported cluster).
- setpoint-raise-lower
- get-relay-status-log

Test log
[TC-TSTAT-3.2.txt](https://github.com/project-chip/connectedhomeip/files/8258720/TC-TSTAT-3.2.txt)

andy31415, 2022-05-30:
SetPointRaiseLower commanad is `M` in the spec, so this is likely cert blocker.

GetRelayStatusLog is optional, so it should not block cert.

if the `SCH` feature is enabled, several other operations become mandatory.

Rajashreekalmane, 2022-05-31:
@cabin15 TSTAT test case works fine with Thermostat app but not in all cluster app.  Could you please use Thermostat app for TSTAT test case in specific.

cjandhyala, 2022-05-31:
@cabin15 this seem to be implemented in all-cluster-app, Can you verify this on all-cluster-app and verify ? Is there a reason you need to verify this on thermostat app, on your platform. 

Can you pls provide platform, app details that you are running this test on ?

cjandhyala, 2022-06-02:
@cabin15 could you verify on latest ToT and file new ticket, since this is old ticket we are closing the issue

closed 2022-06-02 by cjandhyala