project-chip/connectedhomeip #16269: [TE8] TC-MC-9.2 Application Basic Attribute Verification (DUT as Client)

cabin15, 2022-03-16:
Test-Case ID:-[TC-MC-9.2] Application Basic Attribute Verification (DUT as Client)

Problem
Command described on the step 5 of TC-MC-9.2 does not work, but describe chip-tool usage.
Need clarification about verification steps on test plan.

Test log
############### Step 5: ##################

 ./chip-tool applicationlauncher launch-app exampledata 99 unknown 1 0
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] <default-lib> LS_NOT_AN {} Enhanced ACG
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] <default-lib> LS_NOT_AN {} [_LSCheckProvidedTrustedGroups] found group bit mask : 1
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] chip-service MANAGER_SERVICE {} Luna API is called : [sendCommand : 218]
Mar 06 22:11:56 raspberrypi4 ls-hubd[1823]:   ./chip-service applicationlauncher launch-app Application node-id endpoint-id [--commissioner-name] [--ble-adapter] [--timedInteractionTimeoutMs] [--Data]
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] chip-service MANAGER_SERVICE {} Chip Command is called : [makeAndSendCommand : 402]
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] chip-service MANAGER_SERVICE {} Chip Command is called : [sendChipCommand : 389]
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: Run command failure: /usr/src/debug/com.webos.service.matter/1.0.0-te8.3-r1/git/commands/common/Commands.cpp:135: Error 0x0000002F
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] <default-lib> LS_NOT_AN {} Enhanced ACG
Mar 06 22:11:56 raspberrypi4 chip-service[1823]: [] [pmlog] <default-lib> LS_NOT_AN {} [_LSCheckProvidedTrustedGroups] found group bit mask : 1

DEVELOPER COMMENT)
Example command on excel sheet does not work.
root@raspberrypi4:/home/root# ./chip-tool applicationlauncher launch-app exampledata 99 unknown 1 0
Usage:
  ./chip-tool applicationlauncher launch-app Application node-id endpoint-id [--commissioner-name] [--ble-adapter] [--timedInteractionTimeoutMs] [--Data]
root@raspberrypi4:/home/root# Mar 06 22:10:49 raspberrypi4 chip-tool[1741]: Run command failure: ../examples/chip-tool/commands/common/Commands.cpp:135: Error 0x0000002F
root@raspberrypi4:/home/root#

bzbarsky-apple, 2022-05-28:
@cjandhyala This sounds like an issue in the spreadsheet?

cjandhyala, 2022-05-28:
@cabin15 Test plan is updated lately, pls chk with the latest test plans and file the issue in the test plan repo if it is still an issue. Closing this ticket in the SDK repo for now.

closed 2022-05-28 by cjandhyala