project-chip/connectedhomeip #16270: [TC-DM-3.3] Network Commissioning Commands [DUT - Commissioner]

cabin15, 2022-03-16:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-05-21:
@cabin15 Which of the various logs linked here is showing the error?  I tried looking at [chip-tool-TE9.txt](https://github.com/project-chip/connectedhomeip/files/8681080/chip-tool-TE9.txt), [scan-networks.txt](https://github.com/project-chip/connectedhomeip/files/8663159/scan-networks.txt), [scan-network-with-ssid-null.txt](https://github.com/project-chip/connectedhomeip/files/8680945/scan-network-with-ssid-null.txt) but none of them seem to show it?

cabin15, 2022-05-23:
@bzbarsky-apple 
Sorry for confusion.
Please check https://github.com/project-chip/connectedhomeip/files/8258914/TC-DM-3.3.txt.
Below error is shown on the respective log.

2022-03-16T01:11:18.791790Z [417.753360300] user.crit chip-service Received Command Response Status for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0000 Status=0x85
2022-03-16T01:11:18.791835Z [417.753405160] user.crit chip-service Error: IM Error 0x00000585: General error: 0x85

bzbarsky-apple, 2022-05-23:
@cabin15 What actual arguments were sent with ScanNetworks in this case?  The log does not make that clear.

cabin15, 2022-05-23:
Below arguments are used for ScanNetworks.
./chip-tool networkcommissioning scan-networks null 0 12345678 0

bzbarsky-apple, 2022-05-25:
>  ./chip-tool networkcommissioning scan-networks null 0 12345678 0

Those are not valid argument for that command, at least not recently...  If I try to run that, chip-tool just errors out.

Can you please retest whether this is a problem on a recent build?

cabin15, 2022-05-27:
@bzbarsky-apple 
Thank you for your opinion.

I tested scan-networks command on the latest ToT. INVALID_COMMAND is not shown anymore, but shows 0 entry on ScanNetworkResponse.
Here is a ticket regarding 0 entry issue on ScanNetworkResponse.
https://github.com/project-chip/connectedhomeip/issues/18901

bzbarsky-apple, 2022-05-28:
@cabin15 Thank you for the follow-up.  Closing this, since the command does not error out, and will follow up in the other issue.

closed 2022-05-28 by bzbarsky-apple