project-chip/connectedhomeip #16271: [TE8] TC-SC-4.2 Commissionable Node Discovery - Commissioner Case [DUT - Commissioner]

cabin15, 2022-03-16:
Test-Case ID:-[TC-SC-4.2] Commissionable Node Discovery - Commissioner Case [DUT - Commissioner]

Problem
Step 5 of the TC-SC-4.2 is not clear for tester to perform.
{code}
By any means, TH adds an unknown key/value pair in the advertised data(e.g. AB=12345) and is put in Commissioning Mode using Open Basic Commissioning Window command
{code}

outdoor-temperature

occupancy

pi-cooling-demand

and so on (Please check with 0x86 error code on attached log)

Test log

bzbarsky-apple, 2022-05-22:
@cjandhyala This sounds like a bug against the test steps, not the SDK?

cjandhyala, 2022-05-28:
@bzbarsky-apple correct. 
@cabin15 Pls file the issue against test plans

closed 2022-05-28 by cjandhyala