project-chip/connectedhomeip #16272: [TE8] TC-MF-1.11 Device exits commissioning mode after 20 failed commission attempts [ECM] [DUT - Commissioner]

cabin15, 2022-03-16:
Test-Case ID:-[MF-1.11] Device exits commissioning mode after 20 failed commission attempts [BCM] [DUT - Commissioner]

Problem
Expected:
After 20 failed commission attempts using invalid passcode, 21st commissioning attempt should fail.

Actual behavior:
After 20 failed commission attempts using invalid passcode, 21st commissioning attempt succeeds.

Test log
[TC-MF-1.11.txt](https://github.com/project-chip/connectedhomeip/files/8258953/TC-MF-1.11.txt)

bzbarsky-apple, 2022-05-22:
@cabin15 Is this still an issue on tip?

woody-apple, 2022-05-26:
Spec Review: We do not believe this a spec compliance issue, but is a bug. @cabin15 does this still happen? Please re-open if it does.

closed 2022-05-26 by woody-apple