project-chip/connectedhomeip #16299: Pi build (both 32b & 64b) fails - chip-tool.main.cpp.o

HeneryH, 2022-03-16:
#### Problem
Build on pi fails.
System: two systems, same result.  32b and 64b systems
Both systems have had the kernel rebuilt with the cryto modules included.  Was needed for Bluez BT Mesh stuff

git clone --recurse-submodules git@github.com:project-chip/connectedhomeip.git
cd connectedhomeip
git submodule update --init
sudo apt-get install git gcc g++ python pkg-config libssl-dev libdbus-1-dev \\n     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \\n     python3-pip unzip libgirepository1.0-dev libcairo2-dev
sudo apt-get install pi-bluetooth avahi-utils  # these were already installed
sudo vi /etc/systemd/system/dbus-fi.w1.wpa_supplicant1.service
sudo vi /etc/wpa_supplicant/wpa_supplicant.conf
sudo reboot now
cd matter/connectedhomeip
source scripts/activate.sh
```

  WELCOME TO...

           ░▓░
           ▓█▓
           ▓█▓                                                     ▒█     ▒█
      ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
       ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
    ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
    ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
     ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
 ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓
 ░██▒░  ▒██▒ ▒██░  ░▒█▓
         ░▓░ ░▓░

  ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20220316 10:38:14 INF Environment passes all checks!

Environment looks good, you are ready to go!

```
gn gen out/host
ninja -C out/host
```
ninja: Entering directory `out/host'
[504/1588] c++ obj/examples/chip-tool/chip-tool.main.cpp.o
FAILED: obj/examples/chip-tool/chip-tool.main.cpp.o
arm-linux-gnueabihf-g++ -MMD -MF obj/examples/chip-tool/chip-tool.main.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -Wconversion -D_REENTRANT -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -Wno-implicit-fallthrough -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCONFIG_USE_SEPARATE_EVENTLOOP=true -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -I../../examples/chip-tool -I../../zzz_generated/chip-tool -I../../src/lib -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/inipp/repo/inipp -Igen/include -Igen/include -Igen/include -I../../zzz_generated -I../../zzz_generated/controller-clusters -I../../third_party/jsoncpp/repo/include -c ../../examples/chip-tool/main.cpp -o obj/examples/chip-tool/chip-tool.main.cpp.o
arm-linux-gnueabihf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[509/1588] c++ obj/zzz_generated/app-commo...ed/libClusterObjects.cluster-objects.cpp.
ninja: build stopped: subcommand failed.
```
Expected result?  No fatal errors
 
#### Proposed Solution
<suggested fix, suggested enhancement>

HeneryH, 2022-03-18:
I just confirmed the same error on a stock 32bit image without the recompilation of the kernel for the crypto modules.

closed 2022-10-09 by stale[bot]

qq8512852, 2023-09-26:
Hi , I met the same problem in my Raspberry Pi 4B with 64 bit Ubuntu server. 
How do you fix it?

HeneryH, 2023-09-26:
I put this project on a back burner because I never got past these errors.

qq8512852, 2023-10-06:
> I put this project on a back burner because I never got past these errors.

I encountered many issues while compiling on the Raspberry Pi. I might have to put it on hold as well.