project-chip/connectedhomeip #16323: OTA Requestor App attempts second query when killing Provider app to abort transfer midway

ukatira, 2022-03-16:
#### Problem

- Delete CHIP KVS folders for both provider and requestor
- Launch and commission both provider and requestor apps
- Add ACL entry for OTA Provider
- Initiate query using AnnounceOTA
- Kill Provider app to abort transfer midway and wait for BDX Timeout

Requestor App attempts second query when killing Provider app to abort transfer midway instead of waiting for 5 min BDX transfer timeout

 #### Proposed Solution
Requestor App should wait for 5 min BDX transfer timeout and initiate query only after this timeout when triggered using AnnounceOTA or Periodic query.

[ota-p.txt](https://github.com/project-chip/connectedhomeip/files/8268951/ota-p.txt)
[ota-r.txt](https://github.com/project-chip/connectedhomeip/files/8268952/ota-r.txt)

hrajendran-samsung, 2022-03-31:
The issue seems to be that the commissioning-callback timer is not stopped properly before issuing a new Queryimage.

hrajendran-samsung, 2022-03-31:
@ukatira Please note that due to the nature of this timer function, the actual time taken to detect a stalled BDX connection might be anywhere in the range of [mTimeout, 2*mTimeout).
mTimeout being 5 minutes.

PR #16888 (2022-04-01): Ensure that the OTA commissioning callback timer is started with the correct function-pointer and object data
#### Problem
Fixes #16323 
The issue was that the OTA-timer in the "commissioning-done-callback" was never properly stopped (when a new QueryImage request is made). This causes the Requestor to not wait atleast "BDXTimeout" seconds before attempting another request.

#### Change overview
Fix the OTA-start timer (in the commissioning-done-callback) is started with the correct callback/object data to ensure that it gets stopped correctly when a new QueryImage call is made.

#### Testing
 [... cut]
closed 2022-04-01 by hrajendran-samsung