project-chip/connectedhomeip #16332: [TC-LVL-3.1, TC-LVL-4.1, TC-LVL-5.1] Current Level attribute is coming wrong

praveenCY, 2022-03-16:
#### Problem
Current Level attribute is coming wrong 

[LIGHT-TC-LVL-5.1-FAIL.txt](https://github.com/project-chip/connectedhomeip/files/8274691/LIGHT-TC-LVL-5.1-FAIL.txt)
[LIGHT-TC-LVL-4.1-FAIL.txt](https://github.com/project-chip/connectedhomeip/files/8274692/LIGHT-TC-LVL-4.1-FAIL.txt)
[LIGHT-TC-LVL-3.1-FAIL.txt](https://github.com/project-chip/connectedhomeip/files/8274693/LIGHT-TC-LVL-3.1-FAIL.txt)

bzbarsky-apple, 2022-05-22:
@praveenCY Is this still happening on tip?

Rajashreekalmane, 2022-09-28:
@praveenCY Issue was resolved in latest Image, Can you please recheck and close the issue.

PFA logs for reference:-
[TC-LVL-3.1, 4.1 AND 5.1 DUT.log.txt](https://github.com/project-chip/connectedhomeip/files/9662357/TC-LVL-3.1.4.1.AND.5.1.DUT.log.txt)
[[TC-LVL-3.1,4.1and 5.1]Ui automated.log](https://github.com/project-chip/connectedhomeip/files/9662367/TC-LVL-3.1.4.1and.5.1.Ui.automated.log)

closed 2023-05-01 by stale[bot]