project-chip/connectedhomeip #16362: _GetUserLabelList are not able to impl without memory leak

xylophone21, 2022-03-17:
#### Problem
 * expected behavior
    * PlatformManager->_GetUserLabelList and _GetSupportedLocales asked to return a AttributeList with string in it
 * actual behavior
    * No place to free memory in the list

 #### Proposed Solution
* I do not think PlatformManager is a good place to get and set UserLabels, the feature is an app while the platform is something like a driver
* Add some callback to app like src/app/clusters/channel-server/channel-server.cpp::HandleGetChannelList

xylophone21, 2022-03-17:
Right now, only Linux implied this function, and it hardcode some labels which need not be free. If we imp it on a real device, the problem will be found

xylophone21, 2022-03-18:
```
struct DemoLabel {
    char key[32];
    char value[32];
}

_GetUserLabelList(labelList) {
    for() {
         char * label = malloc(DemoLabel); // this memory can not be free before return, but no place to free later

         ChipLinuxStorage::ReadValueStr("label",label->key);
         somelabel.label = CharSpan::fromCharString(label->key);

         ChipLinuxStorage::ReadValueStr("value",label->value);
         somelabel.value = CharSpan::fromCharString(label->value);

         labelList.add(somelabel);
    }

    return labelList;
}
 ```

yufengwangca, 2022-03-19:
With current implementation, the UserLabelList is populated at device domain, the device need to provide the memory storage to store each user label.

By moving this to app domain, like:
CHIP_ERROR ChannelAttrAccess::ReadChannelListAttribute(app::AttributeValueEncoder & aEncoder, Delegate * delegate)
{
    return delegate->HandleGetChannelList(aEncoder);
}

The app can access the AttributeValueEncoder directly, so it can just directly encode the user label into it, so the app domain need to implement some logic like:
        allocate and populate DeviceLayer::AttributeList<app::Clusters::UserLabel::Structs::LabelStruct::Type, DeviceLayer::kMaxUserLabels> labelList;
        err = aEncoder.EncodeList([&labelList](const auto & encoder) -> CHIP_ERROR {
            for (auto label : labelList)
            {
                ReturnErrorOnFailure(encoder.Encode(label));
            }
            return CHIP_NO_ERROR;
        });
        free labelList;

closed 2022-04-01 by yufengwangca