project-chip/connectedhomeip #16376: CASE session reestablishment after power cycling a device

dtodor, 2022-03-17:
When a device is restarted, the CASE session on the controller side is not reestablished. This leads to messages being dropped on the device side:

Controller output:
```
[1647530334.566544][8570:8571] CHIP:DMG: Subscription Liveness timeout with SubscriptionID = 0xD3B80EBB2EA22BE3, Peer = 01:000000000000045A
[1647530334.566690][8570:8571] CHIP:DMG: Computing Resubscribe policy: attempts 0, max wait time 0 ms, selected wait time 0 ms
[1647530334.566789][8570:8571] CHIP:DMG: Will try to Resubscribe to 01:000000000000045A at retry index 0 after 0ms
[1647530334.566825][8570:8571] CHIP:DMG: MoveToState ReadClient[0x71d01f28]: Moving to [      Idle]
[1647530334.567738][8570:8571] CHIP:IN: Prepared secure message 0x560c1124 to 0x000000000000045A (1)  of type 0x3 and protocolId (0, 1) on exchange 20228i with MessageCounter:8303537.
[1647530334.567972][8570:8571] CHIP:IN: Sending encrypted msg 0x560c1124 with MessageCounter:8303537 to 0x000000000000045A (1) at monotonic time: 0000000001AE6E6C msec
[1647530334.568539][8570:8571] CHIP:DMG: MoveToState ReadClient[0x71d01f28]: Moving to [AwaitingIn]
[1647530339.571981][8570:8571] CHIP:EM: Retransmitting MessageCounter:8303537 on exchange 20228i Send Cnt 0
[1647530339.572182][8570:8571] CHIP:IN: Sending encrypted msg 0x560c1124 with MessageCounter:8303537 to 0x000000000000045A (1) at monotonic time: 0000000001AE81F8 msec
[1647530339.572716][8570:8571] CHIP:CTL: Device connected callback with null pairing delegate. Ignoring
[1647530339.578723][8570:8571] CHIP:DL: Resolved interface id: 37
[1647530339.581205][8570:8571] CHIP:DL: Mdns: OnGetAddrInfo hostname:EEE055DEA2BC70CE.local.
[1647530339.581572][8570:8571] CHIP:DIS: Node ID resolved for 0x000000000000045A
[1647530339.581645][8570:8571] CHIP:DIS:     Addr 0: [fdac:77c6:6404:a99b:70b0:9c00:85e2:1127]:5540
[1647530339.582957][8570:8571] CHIP:DIS: Lookup clearing interface for non LL address
[1647530339.583071][8570:8571] CHIP:DIS: UDP:[fdac:77c6:6404:a99b:70b0:9c00:85e2:1127%eth0]:5540: new best score: 6
[1647530339.583121][8570:8571] CHIP:DIS: Checking node lookup status after 10 ms
[1647530339.583152][8570:8571] CHIP:DIS: Keeping DNSSD lookup active
[1647530339.773522][8570:8571] CHIP:DIS: Checking node lookup status after 201 ms
[1647530339.773708][8570:8571] CHIP:CTL: Updating device address to UDP:[fdac:77c6:6404:a99b:70b0:9c00:85e2:1127]:5540 while in state 4
[1647530339.773774][8570:8571] CHIP:DIS: Discovery does not require any more timeouts
[1647530341.572616][8570:8571] CHIP:EM: Retransmitting MessageCounter:8303537 on exchange 20228i Send Cnt 1
[1647530341.572798][8570:8571] CHIP:IN: Sending encrypted msg 0x560c1124 with MessageCounter:8303537 to 0x000000000000045A (1) at monotonic time: 0000000001AE89C9 msec
[1647530343.574661][8570:8571] CHIP:EM: Retransmitting MessageCounter:8303537 on exchange 20228i Send Cnt 2
[1647530343.574851][8570:8571] CHIP:IN: Sending encrypted msg 0x560c1124 with MessageCounter:8303537 to 0x000000000000045A (1) at monotonic time: 0000000001AE919B msec
[1647530345.576925][8570:8571] CHIP:EM: Failed to Send CHIP MessageCounter:8303537 on exchange 20228i sendCount: 3 max retries: 3
[1647530346.568097][8570:8571] CHIP:DMG: Time out! failed to receive report data from Exchange: 20228i
[1647530346.568565][8570:8571] CHIP:DMG: Computing Resubscribe policy: attempts 1, max wait time 10000 ms, selected wait time 5520 ms
[1647530346.568669][8570:8571] CHIP:DMG: Will try to Resubscribe to 01:000000000000045A at retry index 1 after 5520ms
[1647530346.568708][8570:8571] CHIP:DMG: MoveToState ReadClient[0x71d01f28]: Moving to [      Idle]
```

Device output:
```
DestAddr : FDAC:77C6:6404:A99B:70B0:9C00:85E2:1127
Dest Port 5540
Payload Length 59
<error > [IN] Data received on an unknown session (LSID=1). Dropping it!
<detail> [IN] UDP Message Received packet nb : 30 with the following data :
SrcAddr : FDAC:77C6:6404:A99B:B230:B6F8:4E5E:D4A2
Src Port : 48899
```

If the controller is restarted, a new CASE session will be established and everything works as expected.

bzbarsky-apple, 2022-07-14:
See https://github.com/project-chip/connectedhomeip/issues/17298

The plan there would cause the first attempt to talk to the device after it restarts to fail, and later ones to succeed again.

bzbarsky-apple, 2023-01-13:
The plan described above is in place now.

closed 2023-01-13 by bzbarsky-apple