project-chip/connectedhomeip #16378: While verifying the "wild card" read for all the cluster for all the attributes,  chiptool responds as Invalid Cluster-Specific error code 0x11.

Rajashreekalmane, 2022-03-17:
#### Problem
While verifying the wildcard read using the below command, towards the end, Chiptool throws an error with a code 0x11, which is not specified anywhere in the spec.
Can you please clarify the error code?
 * expected behavior
 * actual behavior
 * steps to reproduce:-Execution Command:-./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF

 * system configuration
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 2982c0ca0623773327def4cc7180d07f6b959c42 		
Dut: All-Clusters-app
 
PFA screenshot below:-
[image]

Rajashreekalmane, 2022-03-28:
Problem:
While verifying the "wild card" read for all the cluster for all the attributes for specific endpoint, chiptool responds as Invalid Cluster-Specific error code 0x11.

steps to reproduce:-Execution Command:-./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 1

PFA screenshot below:-
[image]

Rajashreekalmane, 2022-03-28:
Problem:-
While verifying the "wild card" read for all the cluster for all the attributes for specific endpoint, chiptool responds as Run command failure( General error: 0x01).

steps to reproduce:-Execution Command:-./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 1

PFA screenshot below:-
[image]

vivien-apple, 2022-04-20:
This error will not happens for real world device. This is because the `all-clusters-app` contains `TestCluster` which has special attributes that are made to fails when they are read.
For example if you do: 
`./out/debug/standalone/chip-tool testcluster read cluster-error-boolean 0x12344321 1` you will notice the `IM Error 0x00000611: Cluster-specific error: 0x11` code that you are mentioning here.

Reading `general-error-boolean ` also generates an error.

closed 2022-04-21 by Rajashreekalmane