project-chip/connectedhomeip #16398: `chip-tool-darwin` allows for an accessory with PID 00000 to be commissioned

liamgonyea, 2022-03-17:
#### Problem
`chip-tool-darwin` commissions an example accessory with a Product ID of '00000'. However this PID is forbidden per the spec `2.5.3. Product Identifier (Product ID, PID)` stating "A value of 0x0000 SHALL NOT be assigned to a product since Product ID = 0x0000 is used for these specific cases:"

Found in Test Steps 7a/b for TC-DD-3.14
[image]

 #### Proposed Solution
PID may not be checked during commissioning process but per the spec, this PID should not be allowed so there may need to be a check for VID and PID during the commissioning process

#### Testing Setup
TE8
[TC-DD-3.14] Commissioning Flow - Manual Pairing Code - Negative scenario [DUT - Commissioner]
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#3314-tc-dd-314-commissioning-flow---manual-pairing-code---negative-scenario-dut---commissioner

Spec Version: master 40e3631, 2022-01-05 11:22:38 -0800: Draft

bzbarsky-apple, 2022-05-22:
@liamgonyea What are the steps to reproduce here?  That is, how does one configure a test app to have PID == 0?

woody-apple, 2022-05-26:
Assigning to @bzbarsky-apple to confirm if this is fixed.

bzbarsky-apple, 2022-05-26:
After https://github.com/project-chip/connectedhomeip/pull/18606 we disallow commissioning if the setup payload claims PID == 0.

If the setup payload is a short manual code, without PID, then it's not clear to me whether any PID checks should be happening.

But certainly the test plan steps linked above work now.

closed 2022-05-26 by bzbarsky-apple