project-chip/connectedhomeip #16420: [TE8] [TC-MF-1.9] Device still exists in  commissioning mode after 20 failed commission attempts [ECM] [DUT - Commissionee]

AntonGrey, 2022-03-18:
#### DUT Device still exists in  commissioning mode after 20 failed commission attempts

 * Expected behavior: After 20 unsuccessful attempts, the 21st shows that you can’t begin PASE (failure at PBKDFRequest, not obtaining a PBKDFResponse).
 * Actual behavior: After 20 unsuccessful attempts, the 21st commissioning succeeds.
 * Steps to reproduce: TC-MF-1.9
 * Test logs [TC-MF-1.9--failure.log](https://github.com/project-chip/connectedhomeip/files/8302231/TC-MF-1.9--failure.log)

AntonGrey, 2022-05-19:
I have verifiy this issue with TE9 SDK issue.
Looks like original issue isn't reproducible. But actual behavior still doesn't match TestPlan's expected outcome.

* Expected behavior: After 20 unsuccessful attempts, the 21st shows that you can’t begin PASE (failure at PBKDFRequest, not obtaining a PBKDFResponse).
* Actual behavior: After 20 unsuccessful attempts, DUT just closes a commissioning window and the 21st commissioning attempt failure with timeout.
[TC-MF-1.9--TH2.log](https://github.com/project-chip/connectedhomeip/files/8729609/TC-MF-1.9--TH2.log)
[TC-MF-1.9--DUT.log](https://github.com/project-chip/connectedhomeip/files/8729610/TC-MF-1.9--DUT.log)
[TC-MF-1.9--TH1.log](https://github.com/project-chip/connectedhomeip/files/8729611/TC-MF-1.9--TH1.log)

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @cjandhyala to follow up to confirm the issue.

woody-apple, 2022-05-19:
Cert Blocker Review: Tagging @bzbarsky-apple in as well.

bzbarsky-apple, 2022-05-19:
The correct behavior per spec is:

> In both concurrent connection commissioning flow and non-concurrent connection commissioning flow, the Commissionee SHALL exit Commissioning Mode after 20 failed attempts.

That means closing the commissioning window, and no longer listening for PASE connections.  So the TE9 observed behavior is correct per spec.

I just looked at the test plan, and the expected outcome in step 5 in https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/multiplefabrics.adoc#tc-mf-1-9-device-exit-commissioning-mode-after-20-failed-commission-attempts-ecm-dut-commissionee is:

> Verify commissioning flow fails since the commissioning window must be closed after step 4.

which also looks fine.  But the expected outcome in step 4 is nonsense.  Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1286 on that.

closed 2022-05-19 by bzbarsky-apple