project-chip/connectedhomeip #16422: [TE8][TC-LVL-2.1, TC-LVL-3.1, TC-LVL-4.1] TH failure: MinLevel value mismatch: expected 0 but got 1

AntonGrey, 2022-03-18:
#### Certification tests TC-LVL-2.1, TC-LVL-3.1, TC-LVL-4.1 expect MinLevel is 0, actually if LT feature is activated than MinLevel is 1. You can look at [Level cluster specification](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/LevelControl.adoc#15-attributes).

 * Expected behavior: According to the [test procedure](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/levelcontrol.adoc#test-procedure-1) certification test should `verify that the DUT response contains an integer with a maximum size of 8 bits.`
 * Actual behavior: DUT expect MinLevel is 0.
 * steps to reproduce: Look to [TC-LVL-2.1](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/levelcontrol.adoc#test-procedure-1), TC-LVL-3.1, TC-LVL-4.1. 
 * Logs: 
[TC-LVL-2.1--failed.log](https://github.com/project-chip/connectedhomeip/files/8302560/TC-LVL-2.1--failed.log)

cjandhyala, 2022-05-18:
@AntonGrey Could you re-test this with latest SDK and confirm ?

AntonGrey, 2022-05-19:
I have tested these test cases with TE9 SDK version.
The original issue isn't reproduced. But TestHarnes doesn't check returned MinLevel value.
```
CHIPTOOL   | 2022-05-18 04:21:12.271046 | [1652872871.520722][30:40] CHIP:TOO:  ***** Test Step 2 : Reads the MinLevel attribute
CHIPTOOL   | 2022-05-18 04:21:12.273068 | 
CHIPTOOL   | 2022-05-18 04:21:12.274949 | [1652872871.520796][30:40] CHIP:DMG: SendReadRequest ReadClient[0xffff74003470]: Sending Read Request
CHIPTOOL   | 2022-05-18 04:21:12.288273 | [1652872871.520940][30:40] CHIP:IN: Prepared secure message 0xaaaae4bcdbc8 to 0x0F180978BC010491 (1)  of type 0x2 and protocolId (0, 1) on exchange 56964i with MessageCounter:1056896.
CHIPTOOL   | 2022-05-18 04:21:12.290245 | [1652872871.520983][30:40] CHIP:IN: Sending encrypted msg 0xaaaae4bcdbc8 with MessageCounter:1056896 to 0x0F180978BC010491 (1) at monotonic time: 0000000000039AD3 msec
CHIPTOOL   | 2022-05-18 04:21:12.303614 | [1652872871.521167][30:40] CHIP:DMG: MoveToState ReadClient[0xffff74003470]: Moving to [AwaitingIn]
CHIPTOOL   | 2022-05-18 04:21:12.316856 | [1652872871.526394][30:40] CHIP:EM: Received message of type 0x5 with protocolId (0, 1) and MessageCounter:13907547 on exchange 56964i
CHIPTOOL   | 2022-05-18 04:21:12.319899 | [1652872871.526465][30:40] CHIP:EM: Found matching exchange: 56964i, Delegate: 0xffff74003470
CHIPTOOL   | 2022-05-18 04:21:12.322414 | [1652872871.526517][30:40] CHIP:EM: Rxd Ack; Removing MessageCounter:1056896 from Retrans Table on exchange 56964i
CHIPTOOL   | 2022-05-18 04:21:12.325831 | [1652872871.526542][30:40] CHIP:EM: Removed CHIP MessageCounter:1056896 from RetransTable on exchange 56964i
CHIPTOOL   | 2022-05-18 04:21:12.328931 | [1652872871.526618][30:40] CHIP:DMG: ReportDataMessage =
CHIPTOOL   | 2022-05-18 04:21:12.330255 | [1652872871.526650][30:40] CHIP:DMG: {
CHIPTOOL   | 2022-05-18 04:21:12.342742 | [1652872871.526672][30:40] CHIP:DMG: 	AttributeReportIBs =
CHIPTOOL   | 2022-05-18 04:21:12.344689 | [1652872871.527001][30:40] CHIP:DMG: 	[
CHIPTOOL   | 2022-05-18 04:21:12.346890 | [1652872871.527032][30:40] CHIP:DMG: 		AttributeReportIB =
CHIPTOOL   | 2022-05-18 04:21:12.349433 | [1652872871.527064][30:40] CHIP:DMG: 		{
CHIPTOOL   | 2022-05-18 04:21:12.351931 | [1652872871.527086][30:40] CHIP:DMG: 			AttributeDataIB =
CHIPTOOL   | 2022-05-18 04:21:12.354448 | [1652872871.527110][30:40] CHIP:DMG: 			{
CHIPTOOL   | 2022-05-18 04:21:12.356126 | [1652872871.527140][30:40] CHIP:DMG: 				DataVersion = 0xc55e9513,
CHIPTOOL   | 2022-05-18 04:21:12.358548 | [1652872871.527161][30:40] CHIP:DMG: 				AttributePathIB =
CHIPTOOL   | 2022-05-18 04:21:12.370474 | [1652872871.527185][30:40] CHIP:DMG: 				{
CHIPTOOL   | 2022-05-18 04:21:12.372807 | [1652872871.527210][30:40] CHIP:DMG: 					Endpoint = 0x1,
CHIPTOOL   | 2022-05-18 04:21:12.386742 | [1652872871.527234][30:40] CHIP:DMG: 					Cluster = 0x8,
CHIPTOOL   | 2022-05-18 04:21:12.388583 | [1652872871.527260][30:40] CHIP:DMG: 					Attribute = 0x0000_0002,
CHIPTOOL   | 2022-05-18 04:21:12.390524 | [1652872871.527281][30:40] CHIP:DMG: 				}
CHIPTOOL   | 2022-05-18 04:21:12.403228 | [1652872871.527307][30:40] CHIP:DMG: 					
CHIPTOOL   | 2022-05-18 04:21:12.404715 | [1652872871.527333][30:40] CHIP:DMG: 					Data = 1, 
CHIPTOOL   | 2022-05-18 04:21:12.406696 | [1652872871.527353][30:40] CHIP:DMG: 			},
CHIPTOOL   | 2022-05-18 04:21:12.419916 | [1652872871.527381][30:40] CHIP:DMG: 			
CHIPTOOL   | 2022-05-18 04:21:12.421435 | [1652872871.527403][30:40] CHIP:DMG: 		},
CHIPTOOL   | 2022-05-18 04:21:12.423283 | [1652872871.527432][30:40] CHIP:DMG: 		
CHIPTOOL   | 2022-05-18 04:21:12.424836 | [1652872871.527453][30:40] CHIP:DMG: 	],
CHIPTOOL   | 2022-05-18 04:21:12.426773 | [1652872871.527484][30:40] CHIP:DMG:
CHIPTOOL   | 2022-05-18 04:21:12.439337 | [1652872871.527509][30:40] CHIP:DMG: 	SuppressResponse = true, 
CHIPTOOL   | 2022-05-18 04:21:12.441156 | [1652872871.527532][30:40] CHIP:DMG: 	InteractionModelRevision = 1
CHIPTOOL   | 2022-05-18 04:21:12.444457 | [1652872871.527620][30:40] CHIP:DMG: }
CHIPTOOL   | 2022-05-18 04:21:12.446635 | [1652872871.527787][30:40] CHIP:TOO: Warning: value type checking is not implemented yet. Expected type: 'uint8'
CHIPTOOL   | 2022-05-18 04:21:12.449282 | [1652872871.527879][30:40] CHIP:EM: Sending Standalone Ack for MessageCounter:13907547 on exchange 56964i
CHIPTOOL   | 2022-05-18 04:21:12.451651 | [1652872871.527978][30:40] CHIP:IN: Prepared secure message 0xffff7bffd9e8 to 0x0F180978BC010491 (1)  of type 0x10 and protocolId (0, 0) on exchange 56964i with MessageCounter:1056897.
CHIPTOOL   | 2022-05-18 04:21:12.453159 | [1652872871.528041][30:40] CHIP:IN: Sending encrypted msg 0xffff7bffd9e8 with MessageCounter:1056897 to 0x0F180978BC010491 (1) at monotonic time: 0000000000039ADA msec
CHIPTOOL   | 2022-05-18 04:21:12.456541 | [1652872871.528210][30:40] CHIP:EM: Flushed pending ack for MessageCounter:13907547 on exchange 56964i
INFO       | 2022-05-18 04:21:12.653128 | Test Step Completed [PASSED]: Reads the MaxLevel attribute
```

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @cjandhyala

cjandhyala, 2022-05-20:
@AntonGrey Could you file ticket on the script repo https://github.com/CHIP-Specifications/chip-test-scripts, closing this issue as there is no SDK issue here

closed 2022-05-20 by cjandhyala