project-chip/connectedhomeip #16423: [TC-GENDIAG-2.1] - Network-fault event doesn't return any value of NetworkFault enum

sowmyassp, 2022-03-18:
#### Problem
Network-fault event doesn't return any value of NetworkFault enum
 * expected behavior : Should list following
[image]

 * actual behavior : doesn't return anything but gives success response(0x00)
 * steps to reproduce :  ./chip-tool generaldiagnostics read-event network-fault-change 1 0
* screenshots, images, logs, etc.
 >
[network_fault.txt](https://github.com/project-chip/connectedhomeip/files/8302576/network_fault.txt)

sowmyassp, 2022-04-26:
@andy31415 But in test-plan it says we need to get network-fault enum when we run network-fault event, so what I feel is test-plan is wrong. Can you please make sure of it.

sankarselvam, 2022-04-27:
@sowmyassp  , If it is the test plan issue, shall we close this ticket and raise it to the Test plan team? 
Please check the spec and update this ticket

sowmyassp, 2022-04-27:
Yes @sankarselvam it is a test-plan issue I will close the issue here and raise it in test-plan repo.