project-chip/connectedhomeip #16425: [TC-I-2.2] Command "Identify" showing failure response

manjunathambiger23, 2022-03-18:
Summary:
               Command "Identify" showing failure response when we execute

Spec screenshot:
[image]

Execution log and screenshot:
[image]
[tc_I_2_2.txt](https://github.com/project-chip/connectedhomeip/files/8302945/tc_I_2_2.txt)

vivien-apple, 2022-03-24:
The log in `tc_I2_2.txt`seems to contains: 
```
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : [1647592697.840092][49706:49714] CHIP:TOO:  **** Test Setup: Device Connected
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : 
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : [1647592697.840181][49706:49714] CHIP:EM: Sending Standalone Ack for MessageCounter:2522027010 on exchange 41824i
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : [1647592697.840439][49706:49714] CHIP:IN: Prepared unauthenticated message 0x7f8a9159a3c0 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 41824i with MessageCounter:1582480643.
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : [1647592697.840498][49706:49714] CHIP:IN: Sending unauthenticated msg 0x7f8a9159a3c0 with MessageCounter:1582480643 to 0x0000000000000000 at monotonic time: 0000000000F8BD69 msec
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : [1647592697.840606][49706:49714] CHIP:EM: Flushed pending ack for MessageCounter:2522027010 on exchange 41824i
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - APP  OUT  : [1647592697.840684][49684:49684] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:1582480643 on exchange 41824r
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - APP  OUT  : [1647592697.840708][49684:49684] CHIP:EM: Found matching exchange: 41824r, Delegate: (nil)
2022-03-18 14:08:18.975 ERROR   14:08:17.840 - TEST OUT  : [1647592697.840712][49706:49714] CHIP:TOO:  ***** Test Step 1 : Send Identify command to DUT, with the identify time field set to 0x003c (60s)
```

Usually tests logs contains a `***** Test Step 0 : Wait for commissionee` which indicates that the target node has been retrieved from the controller side. Can you double check that the test setup is correct ?

manjunathambiger23, 2022-03-24:
@vivien-apple line number: 2278,  you will get the step 0:  about waiting for commissionee
[image]

bzbarsky-apple, 2022-05-28:
@manjunath-grl @cjandhyala How do I reproduce this?

manjunath-grl, 2022-06-15:
@bzbarsky-apple I tried it in TOT the command is working fine.
SHA : c1e1d018820c0fdd51f124f607b9207f31fae2ff

bzbarsky-apple, 2022-06-16:
@manjunath-grl thanks for retesting!

closed 2022-06-16 by bzbarsky-apple