project-chip/connectedhomeip #16430: Modeselect cluster [TC-MOD-3.3] - Needs SDK implementation on OTA update

somu1710, 2022-03-18:
Problem
In mode select cluster test case [TC-MOD-3.3]  SDK needs to  implement Apply update on device .

ukatira, 2022-03-24:
This test is about persistence of attribute value post an OTA Update but has nothing to do with OTA process itself.

vivien-apple, 2022-03-25:
So, if the OTA update process works, then there is nothing left for me to do ?
@somu1710 When you say "is not able to apply the OTA update" can you provide more informations on what is the issue exactly ?

somu1710, 2022-03-25:
@vivien-apple could u tell me how could i apply the OTA update on device.

vivien-apple, 2022-03-28:
> @vivien-apple could u tell me how could i apply the OTA update on device for modeselect cluster.

@carol-apple  @isiu-apple  Do you have instructions about how to perform an OTA update for a given device ?

@somu1710  Per my understanding this is completely irrelevant to the `ModeSelect` cluster. This is a just an OTA update for the whole device.

Survensa, 2022-03-29:
@vivien-apple and @carol-apple , As per the test plan, we need to reboot the DUT and verify the mode after it reboots.

CUrrently, if we reboot or reinitiate the CHIP Sample DUT, it requires a repairing to read the mode attribute. Hence we filed this ticket to soft reboot the DUT to read the attribute without repairing.

This ticket needs to be assigned to general DUT support to add the soft reboot.

vivien-apple, 2022-04-04:
This is surprising that you need to repair. Why is that ? 
Fir example in YAML there are test step that are rebooting the DUT but it does not requires a repairing and locally I’m killing the server very often without the need to repair. Could it be a misconfiguration ?

carol-apple, 2022-04-04:
Talked with Chaitanya to clear up the ask here:

```
we need get OTA update to working to run this TC, we dont need to build anything for modselect cluster w.r.t OTA.
```

I don't believe there is anything specific in the SDK that needs to be provided for this issue so @somu1710, I think this issue should be closed as the TC you need to run just depends on verification that OTA update works (which is currently being verified).

somu1710, 2022-04-05:
> This is surprising that you need to repair. Why is that ? Fir example in YAML there are test step that are rebooting the DUT but it does not requires a repairing and locally I’m killing the server very often without the need to repair. Could it be a misconfiguration ?

Actually we cant able read/write the commands after reboot the dut without repairing the dut and controller.

somu1710, 2022-04-05:
TC-MOD-3.2 also the same issue we are facing that on reboot the dut. In step no 3a

[image]

cjandhyala, 2022-04-13:
@somu1710 why cant you reboot the device and read the attribute ?

somu1710, 2022-04-13:
If i reboot the device it requires pairing of dut and controller then only I able to read the attribute.

vivien-apple, 2022-04-21:
> If i reboot the device it requires pairing of dut and controller then only I able to read the attribute.could i do like this?

This is the issue. You should not need to repair the device. Do I understand correctly and the device requires re-pairing wether or not there is an OTA update ?

somu1710, 2022-04-21:
yeah I able to do OTA software update

closed 2022-04-21 by somu1710