project-chip/connectedhomeip #16432: [TE8] [TC-GC-1.3] ArmFailSafe command doesn't fail in case if other commissioning process is in progress.

AntonGrey, 2022-03-18:
When TH1’s commissioning  process is in progress, TH2 sends ArmFailSafe command to the same DUT.  As a result, the DUT is expected to respond with error code 0x0004 BUSY. But it doesn't happen, command response with code 0x0 SUCCESS

 * expected behavior: Response with 0x0004 BUSY code
 * actual behavior: Response with 0x0 code
 * steps to reproduce: TC-GC-1.3
 * logs: 
 >
[TC-GC-1.3--failure.log](https://github.com/project-chip/connectedhomeip/files/8304537/TC-GC-1.3--failure.log)

cjandhyala, 2022-05-18:
@AntonGrey do you still see the issue ?

vivien-apple, 2022-05-19:
So looking at the log, the set of commands involved are:
On TH1: `./chip-tool generalcommissioning arm-fail-safe 60 0 60 1 0`
On TH2: `./chip-tool generalcommissioning arm-fail-safe 950 1 100 1 0`

Which, per my understanding, implies:
 * Commissioning window is not opened
 * On the first command, the fail-safe-timer is not armed. though it is armed on the second command
 * Commands are sent over a CASE session
 * Commands are sent using the "alpha" fabric. Both commands has the same fabric.

If my understanding is correct, this means that the set of commands falls under the current condition:
```
If ExpiryLengthSeconds is non-zero and the fail-safe timer was currently armed, and the accessing Fabric matches the fail-safe context’s associated Fabric, then the fail-safe timer SHALL be re-armed to expire in ExpiryLengthSeconds.
```

In this case, it seems like the correct behaviour or did I misunderstood the scenario ?

AntonGrey, 2022-05-19:
Looks like TC-GC-1.3 was removed from the Test Plan.
@vivien-apple thank you for explanation.  You are definitely right. 
Originally, I was confused by the Verification steps document.  There were steps with mistakes. 
Actually, I have performed TC-GC-2.3 which looks very close to original TC-GC-1.3. This test was successful.

closed 2022-05-19 by AntonGrey