project-chip/connectedhomeip #16434: [TC-DD-3.4] - Step 4 says TH must find the DUT Commissioner which fails to do so

sowmyassp, 2022-03-18:
#### Problem
TH is instructed to start the commissioning procedure with the DUT found at step 2 where tv-casting-app is failing to do so
 * expected behavior : TH is instructed to start the commissioning procedure with the DUT found at step 2
 * actual behavior : But DUT is not found as tv-casting -app is failing to do so 
 * steps to reproduce : step 1) sudo ./chip-tv-app
                                    step 2) sudo ./chip-tv-casting-app
[image]

 * screenshots, images, logs, etc.
 >
[tv-casting-app.txt](https://github.com/project-chip/connectedhomeip/files/8304849/tv-casting-app.txt)

chrisdecenzo, 2022-03-22:
I think this is already fixed in top of tree. What branch are you building tv-casting-app from?

sowmyassp, 2022-03-23:
@chrisdecenzo I build tv-app using Linux platform on master branch

closed 2022-04-05 by sowmyassp